carp-dk/carp.sensing-flutter #4: Lack of permissions makes the app crash

thomasnilsson, 2018-10-29:
In `carp_mobile_sensing_app` the study will start before the user is prompted for permissions, such as location, microphone and so on, which leads to the app crashing.

We need a way to await the permission before starting any probes - not sure if this should be done on the "app" side or the domain side. 

We can probably just use this library for checking permissions https://pub.dartlang.org/packages/simple_permissions

Whathecode, 2018-10-29:
This should be done in both. The domain should know about the fact that requested sensors do not necessarily work immediately. They need some 'configuration'.

bardram, 2018-12-30:
The app also crashes if the Firebase CARP plugin isn't configured according to the guidelines stated at;

https://github.com/cph-cachet/carp.sensing-flutter/tree/master/carp_firebase_backend

The Google Firebase plugin just crash without any warning, so it's hard to see that the root cause is Firebase.

I'm looking into these issues.

bardram, 2019-10-18:
This is an old issue, and most of this has been solved in later versions.

closed 2019-10-18 by bardram