carp-dk/carp.sensing-flutter #16: screenStateEvent Exception

DARdk, 2019-02-11:
running carp_mobile_sensing version: 0.3.2
iOS 9.0 

In console I get following warning: 
```
EXCEPTION CAUGHT BY SERVICES LIBRARY
The following MissingPluginException was thrown while activating platform stream on channel
screenStateEvents:
MissingPluginException (No implementation found for method listen on channel screenStateEvents)
```
Look in: platform_channel.dart:300:7

nightscape, 2019-02-23:
I had the same error and was able to fix it by adding an explicit `screen_state: "0.0.1"` dependency to `pubspec.yaml` or alternatively locking it in `pubspec.lock`:
```
screen_state:
    dependency: "direct main"
    description:
      name: screen_state
      url: "https://pub.dartlang.org"
    source: hosted
    version: "0.0.1"
```

thomasnilsson, 2019-02-24:
Will fix this tonight

closed 2019-04-30 by thomasnilsson