carp-dk/carp.sensing-flutter #17: MissingPluginException getUsage

DARdk, 2019-02-12:
```Carp_mobile_sensing vs 0.3.2```
When starting the AppUsageProbe on iOS I get the following error: 
```
flutter: starting Instance of 'AppUsageProbe'
Unhandled Exception: 'file:///Users/a... line 20 pos 15: 'googleAppID !=null is not true
```
It doesnt seem like it can initiate this probe

thomasnilsson, 2019-02-13:
The specific probe has been fixed now, as in it no longer tries to invoke a method which does not exist on the iOS side. I will be updating the other plugins as well.

I managed to run the project on iOS by

* Changing to the legacy build system ([link](https://github.com/flutter/flutter/issues/20685#issuecomment-413524718)).
* Changing the podspec version to ios 10.0 ([link](https://stackoverflow.com/questions/51784066/flutter-ios-cocoapods-could-not-find-compatible-versions-for-pod-simple-permi))

DARdk, 2019-02-13:
Perfect! 
I managed to run the project on iOS as well. 
I also had to change to the legacy build system.
However, I was able to run with podspec version ios 9.0. 
I just had to change the dependency of simple_permission within the ios/.symlinks/plugins/simple_permission/simple_permission.

closed 2019-04-30 by thomasnilsson