carp-dk/carp.sensing-flutter #18: [mobile sensing app] ActivityRecognition header file not found

thomasnilsson, 2019-02-13:
It seems like the ActivityRecognition plugin is not properly linked in XCode since `#import <activity_recognition/ActivityRecognitionPlugin.h>` results in the following error: 

`'activity_recognition/ActivityRecognitionPlugin.h' file not found`

thomasnilsson, 2019-04-30:
The GitHub repo for [activity recognition](https://pub.dartlang.org/packages/activity_recognition) has been taken down, do we continue using it?

bardram, 2019-06-04:
An alternative activity recognition plugin has been made available:

* `activity_recognition_alt: ^0.1.7`

This is based on a fork/download of the original plugin.

The question is whether we want to maintain our own?

bardram, 2019-06-04:
Maybe keep this open until you @thomasnilsson has tested it on iOS....

bardram, 2019-09-08:
Have tested it with `activity_recognition_alt` and it comes with an error when building the pods (or whatever they are called on iOS):

```
'activity_recognition_alt/activity_recognition-Swift.h' file not found
```

I fixed this by changing it to:

```
'activity_recognition_alt/activity_recognition_alt-Swift.h' file not found
```

(adding `_alt` to the filename).

bardram, 2019-09-08:
On iOS, however, it throws an exception from the `activity_recognition_alt` plugin:

```
flutter: Another exception was thrown: MissingPluginException(No implementation found for method listen on channel activity_recognition/activityUpdates)
[VERBOSE-2:ui_dart_state.cc(148)] Unhandled Exception: MissingPluginException(No implementation found for method startActivityUpdates on channel activity_recognition/activities)
#0      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:314:7)
<asynchronous suspension>
#1      _ActivityChannel.startActivityUpdates (package:activity_recognition_alt/channel/activity_channel.dart:26:14)
#2      new _ActivityChannel (package:activity_recognition_alt/channel/activity_channel.dart:16:42)
#3      ActivityRecognitionAlt._activityChannel (package:activity_recognition_alt/activity_recognition_alt.dart:22:35)
#4      ActivityRecognitionAlt._activityChannel (package:activity_recognition_alt/activity_recognition_alt.dart:22:16)
#5      ActivityRecognitionAlt.activityUpdates (package:activity_recognition_alt/activity_recognition_alt.dart:20:48)
```

mvictoras, 2019-09-25:
Any updates on this? I am getting the same error:
`'activity_recognition_alt/activity_recognition-Swift.h' file not found`

and changing it to `'activity_recognition_alt/activity_recognition_alt-Swift.h'` does not seem to fix it.

mvictoras, 2019-09-25:
I stand corrected. Changing it to `'activity_recognition_alt/activity_recognition_alt-Swift.h'` does fix the issue, however I am now having a new issue:

`fatal error:
    'flutter_blue/FlutterBluePlugin.h' file not found
    #import <flutter_blue/FlutterBluePlugin.h>`

mvictoras, 2019-09-27:
@bardram @thomasnilsson any updates on this?

mvictoras, 2019-09-27:
I was able to fix it by copying the missing headers:
cp flutter/.pub-cache/hosted/pub.dartlang.org/flutter_blue-0.6.2/ios/Classes/Flu ./carp.sensing-flutter/carp_mobile_sensing_app/build/ios/Debug-iphoneos/flutter_blue/flutter_blue.framework/Headers/

and wget https://raw.githubusercontent.com/pauldemarco/flutter_blue/master/ios/gen/Flutterblue.pbobjc.h

bardram, 2019-10-25:
Hi @mvictoras - we're right now doing some thorough testing on iOS and will get back on this. We are, however, a little held back with some Flutter issues on iOS 13, like [#41133](https://github.com/flutter/flutter/issues/41133).

bardram, 2019-11-06:
When testing on iOS in Xcode, still get the missing plugin exception:

```
2019-11-06 11:08:59.958529+0100 Runner[1955:543883] flutter: ══╡ EXCEPTION CAUGHT BY SERVICES LIBRARY ╞══════════════════════════════════════════════════════════
2019-11-06 11:08:59.958612+0100 Runner[1955:543883] flutter: The following MissingPluginException was thrown while activating platform stream on channel
2019-11-06 11:08:59.958635+0100 Runner[1955:543883] flutter: activity_recognition/activityUpdates:
2019-11-06 11:08:59.958662+0100 Runner[1955:543883] flutter: MissingPluginException(No implementation found for method listen on channel
2019-11-06 11:08:59.958681+0100 Runner[1955:543883] flutter: activity_recognition/activityUpdates)
2019-11-06 11:08:59.958697+0100 Runner[1955:543883] flutter:
2019-11-06 11:08:59.958720+0100 Runner[1955:543883] flutter: When the exception was thrown, this was the stack:
2019-11-06 11:08:59.958744+0100 Runner[1955:543883] flutter: #0      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:314:7)
2019-11-06 11:08:59.958950+0100 Runner[1955:543883] flutter: <asynchronous suspension>
2019-11-06 11:08:59.959188+0100 Runner[1955:543883] flutter: #1      EventChannel.receiveBroadcastStream.<anonymous closure> (package:flutter/src/services/platform_channel.dart:504:29)
2019-11-06 11:08:59.959644+0100 Runner[1955:543883] flutter: <asynchronous suspension>
2019-11-06 11:08:59.959674+0100 Runner[1955:543883] flutter: #6      _ActivityChannel.startActivityUpdates (package:activity_recognition_alt/channel/activity_channel.dart:24:10)
2019-11-06 11:08:59.959698+0100 Runner[1955:543883] flutter: #7      new _ActivityChannel (package:activity_recognition_alt/channel/activity_channel.dart:16:42)
2019-11-06 11:08:59.960057+0100 Runner[1955:543883] flutter: #8      ActivityRecognitionAlt._activityChannel (package:activity_recognition_alt/activity_recognition_alt.dart:22:35)
2019-11-06 11:08:59.960288+0100 Runner[1955:543883] flutter: #9      ActivityRecognitionAlt._activityChannel (package:activity_recognition_alt/activity_recognition_alt.dart:22:16)
2019-11-06 11:08:59.960405+0100 Runner[1955:543883] flutter: #10     ActivityRecognitionAlt.activityUpdates (package:activity_recognition_alt/activity_recognition_alt.dart:20:48)
2019-11-06 11:08:59.960514+0100 Runner[1955:543883] flutter: #11     ActivityProbe.onInitialize (package:carp_context_package/activity_probe.dart:16:28)
2019-11-06 11:08:59.960621+0100 Runner[1955:543883] flutter: #13     ActivityProbe.onInitialize (package:carp_context_package/activity_probe.dart:13:28)
2019-11-06 11:08:59.960727+0100 Runner[1955:543883] flutter: #14     _CreatedState.initialize (package:carp_mobile_sensing/runtime/probes.dart:241:19)
2019-11-06 11:08:59.960848+0100 Runner[1955:543883] flutter: #16     _CreatedState.initialize (package:carp_mobile_sensing/runtime/probes.dart:238:26)
2019-11-06 11:08:59.960970+0100 Runner[1955:543883] flutter: #17     AbstractProbe.initialize (package:carp_mobile_sensing/runtime/probes.dart:152:19)
2019-11-06 11:08:59.961080+0100 Runner[1955:543883] flutter: #19     AbstractProbe.initialize (package:carp_mobile_sensing/runtime/probes.dart:148:26)
```

We may need to roll our own activity recognition plugin?

svoida, 2019-11-15:
Also having trouble compiling for ios. Seems like this might be related to https://github.com/pauldemarco/flutter_blue/issues/425, which is currently unresolved.

thomasnilsson, 2020-03-25:
We have now created our own fork of the `activity_recognition_alt` plugin. It can be found here: https://pub.dev/packages/activity_recognition_flutter#-changelog-tab-

closed 2020-03-25 by thomasnilsson