carp-dk/carp.sensing-flutter #23: Audio recording fails

bardram, 2019-04-17:
When trying to record sound in the CARP Mobile Sensing App, there is an exception from Flutter Sound

```
D/FlutterSoundPlugin( 1311): startRecorder
I/MediaRecorderJNI( 1311): setOutputFile
E/MediaRecorder( 1311): output file has already been set
E/FlutterSoundPlugin( 1311): Exception: 
E/FlutterSoundPlugin( 1311): java.lang.IllegalStateException
E/FlutterSoundPlugin( 1311): 	at android.media.MediaRecorder._setOutputFile(Native Method)
E/FlutterSoundPlugin( 1311): 	at android.media.MediaRecorder.prepare(MediaRecorder.java:1266)
E/FlutterSoundPlugin( 1311): 	at com.dooboolab.fluttersound.FlutterSoundPlugin.startRecorder(FlutterSoundPlugin.java:161)
E/FlutterSoundPlugin( 1311): 	at com.dooboolab.fluttersound.FlutterSoundPlugin.lambda$onMethodCall$0$FlutterSoundPlugin(FlutterSoundPlugin.java:65)
E/FlutterSoundPlugin( 1311): 	at com.dooboolab.fluttersound.FlutterSoundPlugin$$Lambda$0.run(Unknown Source:8)
E/FlutterSoundPlugin( 1311): 	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:457)
E/FlutterSoundPlugin( 1311): 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
E/FlutterSoundPlugin( 1311): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
E/FlutterSoundPlugin( 1311): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
E/FlutterSoundPlugin( 1311): 	at java.lang.Thread.run(Thread.java:764)
I/flutter ( 1311): >> {"latitude":{"value":55.7943012,"unit":"deg"},"longitude":{"value":12.4461802,"unit":"deg"},"positioning_system":"GPS"}
I/flutter ( 1311): >> {"id":"cefcb500-5970-11e9-ad94-5f62322a1314","timestamp":"2019-04-17T09:54:36.239681Z","mean_lux":106.66566265060241,"std_lux":1.5714782333849235,"min_lux":99,"max_lux":109}
E/flutter ( 1311): [ERROR:flutter/lib/ui/ui_dart_state.cc(148)] Unhandled Exception: Exception: Recorder already stopped.
E/flutter ( 1311): null
I/flutter ( 1311): >> {"id":"cf99df60-5970-11e9-862c-79cf2cefd498","timestamp":"2019-04-17T09:54:37.269941Z","message":"AudioProbe Error: Exception: Recorder already stopped."}

```

thomasnilsson, 2019-04-17:
Will look at it this weekend

thomasnilsson, 2019-04-21:
It is a problem with the flutter_sound package, made by a 3rd party.

I managed to replicate the ´output file has already been set´-error using the in a new project (only using the ´flutter_sound´ package), and it boils down to starting- and stopping the recording too fast. (see https://github.com/dooboolab/flutter_sound/issues/53)

I tried configuring the duration of the ambient recordings to last longer than 2 seconds, but the problem does not seem to go away. Are we starting and stopping the recording somewhere in the code?

bardram, 2019-04-28:
I've fixed it now. A few updates to the audio probe. But the main problem is that the AUDIO and the NOISE probe *cannot* run at the *same* time -- they conflict. Probably because they use the same underlying audio stream. But closing this issue, and creating a new.

closed 2019-04-28 by bardram