carp-dk/carp.sensing-flutter #24: Noise and Audio probe cannot run at the same time

bardram, 2019-04-28:
It seems like the access to the microphone is "taken" by the Noise probe.  For example in the example below, first the noise probe collect data (for 2 secs) and work fine and returns a noise recording. Then, when the audio probe wants to record, it get and error in starting the recording.

````
I/flutter ( 5788): Starting Instance of 'NoiseProbe'
I/MediaRecorderJNI( 5788): setup
I/flutter ( 5788): Sensing started ...
I/MediaRecorderJNI( 5788): setAudioSource(1)
I/MediaRecorderJNI( 5788): setAudioEncoder(0)
I/MediaRecorderJNI( 5788): setOutputFile
I/MediaRecorderJNI( 5788): prepare
I/MediaRecorderJNI( 5788): start
D/NoiseMeterPlugin( 5788): listen()
I/flutter ( 5788): >> {"id":"fece0e90-5970-11e9-f9a7-a331c1237735","timestamp":"2019-04-28T11:30:16.184358Z","mean_decibel":34.23076923076923,"std_decibel":5.984859266008269,"min_decibel":0,"max_decibel":52}
I/flutter ( 5788): Noise - mean: 34.23076923076923, std: 5.984859266008269, min: 0, max: 52
D/FlutterSoundPlugin( 5788): startRecorder
I/MediaRecorderJNI( 5788): setup
I/MediaRecorderJNI( 5788): setAudioSource(1)
I/MediaRecorderJNI( 5788): setAudioEncoder(3)
I/MediaRecorderJNI( 5788): setParameter()
I/MediaRecorderJNI( 5788): setOutputFile
I/MediaRecorderJNI( 5788): prepare
I/MediaRecorderJNI( 5788): start
E/MediaRecorder( 5788): start failed: -38
E/FlutterSoundPlugin( 5788): Exception: 
E/FlutterSoundPlugin( 5788): java.lang.IllegalStateException
E/FlutterSoundPlugin( 5788): 	at android.media.MediaRecorder._start(Native Method)
E/FlutterSoundPlugin( 5788): 	at android.media.MediaRecorder.start(MediaRecorder.java:1310)
E/FlutterSoundPlugin( 5788): 	at com.dooboolab.fluttersound.FlutterSoundPlugin.startRecorder(FlutterSoundPlugin.java:162)
E/FlutterSoundPlugin( 5788): 	at com.dooboolab.fluttersound.FlutterSoundPlugin.lambda$onMethodCall$0$FlutterSoundPlugin(FlutterSoundPlugin.java:65)
E/FlutterSoundPlugin( 5788): 	at com.dooboolab.fluttersound.FlutterSoundPlugin$$Lambda$0.run(Unknown Source:8)
E/FlutterSoundPlugin( 5788): 	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:457)
E/FlutterSoundPlugin( 5788): 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
E/FlutterSoundPlugin( 5788): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
E/FlutterSoundPlugin( 5788): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
E/FlutterSoundPlugin( 5788): 	at java.lang.Thread.run(Thread.java:764)
````

bardram, 2019-04-28:
@thomasnilsson - maybe the noise plugin does not release the microphone?

thomasnilsson, 2019-04-28:
Will have a look at the underlying implementation

thomasnilsson, 2019-04-30:
It does not seem possible to run two plugins which use the same microphone, at the same time. 

However there is no problem running them in an interleaving fashion, where only one plugin is using the microphone at one time.

bardram, 2019-06-04:
We have some more fundamental problems w. the audio plugin -- have raised an issue:

* https://github.com/dooboolab/flutter_sound/issues/93

bardram, 2019-11-06:
A new release of flutter_sound fixing #93 above have been released and I've updated the CARP package. However, still have not tested if the noise and audio probe can be used in the same study, if taking care of not running at the same time (interleaving).

bardram, 2021-08-03:
With the upgrade to `carp_audio_package` v. 0.30.0 using `flutter_sound: ^8.1.4` for audio recording, this seems to be working now.

closed 2021-08-03 by bardram