carp-dk/carp.sensing-flutter #26: How to pass parameters to a probe?

nightscape, 2019-05-14:
I am currently trying to extend CARP with a Shimmer sensor that is connected via Bluetooth.
The MAC address of the device will only be known at runtime, after the user has selected the device from a list of all Bluetooth devices.
Now the question is, how do I pass a `String macAddress` parameter from the UI to the `ShimmerProbe` where it will be used to connect to the device.
My intuition is that the `ShimmerMeasure` would be the right place to store the parameter and then read it in `ShimmerProbe#onInitialize(Measure measure)`, but the `ShimmerMeasure` gets instantiated in getter methods which don't accept any parameters.
The only option I see right now is to give the `ShimmerSamplingPackage` a `macAddress` field and initialize that in the constructor. Would that be the way to go?

As a simplified use example, how would the `WeatherMeasure` work if the token was not known at compile time, but needs to be specified by the user at runtime?

Whathecode, 2019-05-16:
At the risk of mentioning things which are out of scope here, I can provide input on what the intended design of the overarching CARP architecture is, which is larger than just Flutter sensing library. Mostly for @bardram or others working on the implementation of this library.

According to the domain model, such data should be stored in a `DeviceRegistration` which gets registered to a specific `DeviceDescriptor` as originally specified in the `Protocol` which describes the entire study: https://github.com/cph-cachet/carp.core-kotlin/blob/develop/carp.deployment.core/src/commonMain/kotlin/dk/cachet/carp/deployment/application/DeploymentManager.kt#L55

How this `DeviceRegistration` gets passed on the the Flutter sensing library running on e.g., a smartphone, is not yet specified. But, the idea was to simply use the same object instance as also used on server-side. This is essentially a _shared kernel_ (DDD speak) in between two separate subsystems: the `Deployment` manager and `Client` runtime.

The scope of `Measure` instances should be limited to specifying _what_ should be measured, not how. E.g. a [`PhoneSensorMeasure`](https://github.com/cph-cachet/carp.core-kotlin/blob/develop/carp.protocols.core/src/commonMain/kotlin/dk/cachet/carp/protocols/domain/tasks/measures/PhoneSensorMeasure.kt) specifies which of the built-in sensors to measure when a `Task` is triggered. But, such definitions are not fully aligned yet between the Flutter and Kotlin implementation and are still in flux.

@bardram is also looking into seperating the Flutter sensing library entirely from the CARP's protocol definitions etc. so it may be used individually. The `DeviceRegistration` described above would at some point simply need to be passed down to the independent Flutter sensing library. In this case, I believe this information should be part of the 'probe'.

bardram, 2019-06-04:
@nightscape -- sorry for this late answer to your question. But I've updated the documentation on the wiki to describe how to use the [lifecycle methods](https://github.com/cph-cachet/carp.sensing-flutter/wiki/3.-Using-CARP-Mobile-Sensing#lifecycle-methods) including how to update a `Measure` and then use the `hasChanged()` method to let the probe know, that its measure (configuration) has changed.

I hope this helps. Remember to use the latest version 0.4.0.

closed 2019-06-04 by bardram