carp-dk/carp.sensing-flutter #44: carp_mobile_sensing_app will not build on Android

willhaslett, 2019-12-10:
### Problem
After opening the carp_mobile_sensing app in Android Studio (commit `667fcd9fd89ff36413de18f145f21b17c4837029` on 2019-12-09), and getting packages, this example app fails to build.

### Steps to reproduce
1. Open the carp_mobile_sensing_app in Android Studio
2. Get any needed packages
3. Attempt build/run on an Android target

### logcat error (x3)
```
12-10 17:36:22.914  8282  8324 W ErrorReporter: reportError [type: 211, code: 393244, bug: 0]: errorCode: 393244, engine: 0
12-10 17:36:22.915  8282  8324 I MicroDetector: Keeping mic open: false
12-10 17:36:22.915  8282  8324 I MicroDetectionWorker: #onError(false)
```

### Gradle output
```
Launching lib/main.dart on Android SDK built for x86 in debug mode...
Initializing gradle...
Resolving dependencies...
* Error running Gradle:
ProcessException: Process "/Users/will/code/carp.sensing-flutter/carp_mobile_sensing_app/android/gradlew" exited abnormally:

> Configure project :app
WARNING: API 'variant.getMergeAssets()' is obsolete and has been replaced with 'variant.getMergeAssetsProvider()'.
It will be removed at the end of 2019.
For more information, see https://d.android.com/r/tools/task-configuration-avoidance.
To determine what is calling variant.getMergeAssets(), use -Pandroid.debug.obsoleteApi=true on the command line to display more information.
WARNING: API 'variantOutput.getProcessResources()' is obsolete and has been replaced with 'variantOutput.getProcessResourcesProvider()'.
It will be removed at the end of 2019.
For more information, see https://d.android.com/r/tools/task-configuration-avoidance.
To determine what is calling variantOutput.getProcessResources(), use -Pandroid.debug.obsoleteApi=true on the command line to display more information.

> Configure project :activity_recognition_alt
WARNING: API 'variant.getJavaCompile()' is obsolete and has been replaced with 'variant.getJavaCompileProvider()'.
It will be removed at the end of 2019.
For more information, see https://d.android.com/r/tools/task-configuration-avoidance.
To determine what is calling variant.getJavaCompile(), use -Pandroid.debug.obsoleteApi=true on the command line to display more information.

> Configure project :flutter_blue
The com.google.protobuf plugin was already applied to the project: :flutter_blue and will not be applied again after plugin: android-library
         *********************************************************
WARNING: This version of flutter_blue will break your Android build if it or its dependencies aren't compatible with AndroidX.
         See https://goo.gl/CP92wY for more information on the problem and how to fix it.
         This warning prints for all Android build failures. The real root cause of the error may be unrelated.
         *********************************************************
  Command: /Users/will/code/carp.sensing-flutter/carp_mobile_sensing_app/android/gradlew app:properties

Finished with error: Please review your Gradle project setup in the android/ folder.
```

### Flutter doctor -v
```
[✓] Flutter (Channel stable, v1.9.1+hotfix.6, on Mac OS X 10.14.6 18G1012, locale en-US)
    • Flutter version 1.9.1+hotfix.6 at /Users/will/flutter
    • Framework revision 68587a0916 (3 months ago), 2019-09-13 19:46:58 -0700
    • Engine revision b863200c37
    • Dart version 2.5.0

[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
    • Android SDK at /Users/will/Library/Android/sdk
    • Android NDK location not configured (optional; useful for native profiling support)
    • Platform android-29, build-tools 29.0.2
    • Java binary at: /Applications/Android Studio.app/Contents/jre/jdk/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 1.8.0_202-release-1483-b49-5587405)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS (Xcode 11.2.1)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 11.2.1, Build version 11B500
    • CocoaPods version 1.6.2

[✓] Android Studio (version 3.5)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin version 41.1.2
    • Dart plugin version 191.8593
    • Java version OpenJDK Runtime Environment (build 1.8.0_202-release-1483-b49-5587405)

[✓] Connected device (2 available)
    • Android SDK built for x86 • emulator-5554                        • android-x86 • Android 10 (API 29) (emulator)
    • iPhone 11 Pro Max         • AF5B6408-768A-4C8C-A392-25125F77272C • ios         • com.apple.CoreSimulator.SimRuntime.iOS-13-2 (simulator)

• No issues found!
```

### iOS
iOS will not build either, and XCode throws this error:
```
    [!] Automatically assigning platform `ios` with version `9.0` on target `Runner` because no platform was specified. Please specify a platform for this target in your Podfile. See `https://guides.cocoapods.org/syntax/podfile.html#platform`.
```

bardram, 2020-05-01:
We're working on a new releasee of both CAMS and the app. Here we're adressing these things.

willhaslett, 2020-05-01:
@bardam sounds great. We will be keeping an eye out! It feels like there should be an academic Flutter interest group, if there isn't already. Our passive sensor / survey platform at Dartmouth is coming along (we decided to roll our own), and I hope to get the go-ahead to open source it sometime this year. Big Open Science opportunities here!

bardram, 2020-05-24:
Hi @willhaslett - we've released version `0.7.x` of the framework including the app. I hope this helps you compile and use CAMS now. Sorry for taking so long, but had to implement the new `AppTask` model also.

closed 2020-05-27 by bardram