carp-dk/carp.sensing-flutter #51: Issues with downloading dependencies

ottoman91, 2020-02-14:
Not sure if this is the right place to be adding this, but can I have more clarity on how to download all the dependencies for the packages? I have downloaded the repo, and navigated to the location of every module and ran `flutter pub get` . However, when I run this command in the `carp_mobile_sensing_app` directory, I am getting the following errors:

`
! carp_audio_package 0.6.1 from path ../carp_audio_package              
! carp_backend 0.6.0+1 from path ../carp_backend                        
! carp_communication_package 0.6.1+1 from path ../carp_communication_package
! carp_context_package 0.6.4 from path ../carp_context_package          
! carp_esense_package 0.1.4 from path ../carp_esense_package            
! carp_mobile_sensing 0.6.3 from path ../carp_mobile_sensing            
! carp_survey_package 0.6.0 from path ../carp_survey_package            
! carp_webservices 0.3.3+1 from path ../carp_webservices 
` 
I have tried to change the version of the dependencies in the carp_mobile_sensing_app/pubspec.yaml file, but am still getting this error.

Thanks!

bardram, 2020-04-05:
Hi @ottoman91 -- it seems like the version of the app in the master branch have these conflicting issues with package versions. Will look into it and make sure that a version of the app in the master branch is working. Hang on....

bardram, 2020-05-24:
Hi @ottoman91 -- a new version `0.7.x` of the framework and the app is now released. Hope this solved the issue - at least I can compile the example app now.

closed 2020-05-27 by bardram