carp-dk/carp.sensing-flutter #54: Backend: Carp webservice repo no longer available

nstrelow, 2020-04-27:
Currently I see 3 options of saving Study data:
1. CARP Webservice
2. Firebase
3. Locally

Carp webservice links to a github repo, which is no longer available.

Is CARP webservice something which was/will be open source?

What is the recommended way of setting up a CARP system outside of CACHET?

bardram, 2020-05-01:
The CARP backend is available but it is a privacy github repro at the moment. We're not done with a first release of it yet, and we're only using it internally for our own apps. But the plan is to release the CARP backend service at some point. 

But - the whole idea of the CAMS backend architecture is that it should be easy to implement you own -- for example if you want to use your own servers or AWS or ... See this small tutorial on how to do this

https://github.com/cph-cachet/carp.sensing-flutter/wiki/4.-Extending-CARP-Mobile-Sensing#adding-a-new-data-manager

bardram, 2020-05-01:
Will close this for now.

closed 2020-05-01 by bardram

nstrelow, 2020-05-01:
Thanks a lot for the answers on this and other issues and this awesome plugin suite. Looking forward to further development.