carp-dk/carp.sensing-flutter #56: Context: activity_recognition_alt uses incompatible kotlin version

nstrelow, 2020-04-27:
I see you have your own activity_recognition package:
Would be nice, if you could update context to use it:
https://pub.dev/packages/activity_recognition_flutter

When building the app, I get following error:

```
FAILURE: Build failed with an exception.

* What went wrong:
The Android Gradle plugin supports only Kotlin Gradle plugin version 1.3.10 and higher.
The following dependencies do not satisfy the required version:
project ':activity_recognition_alt' -> org.jetbrains.kotlin:kotlin-gradle-plugin:1.2.51
```

bardram, 2020-05-01:
Yep - will be in the coming update.

bardram, 2020-05-27:
@nstrelow - a new version 0.7.x of CAMS has been released with the new activity_recognition_flutter plugin. See

 * https://pub.dev/packages/carp_mobile_sensing
 * https://pub.dev/packages/carp_context_package

closed 2020-05-27 by bardram