carp-dk/carp.sensing-flutter #61: ConditionalSamplingEventTrigger not triggering

MadsVSChristensen, 2020-05-07:
`ConditionalSamplingEventTrigger` does not trigger when a datum with the correct conditions are present. 
Seemingly, the `ConditionalSamplingEventTriggerExecutor` doesn't  listen on the event datum stream, and therefore the `resumeCondition` is never checked.

bardram, 2020-05-20:
I cannot reconstruct the error. It works fine for me. I have been testing with the following study configuration.

```
_study = Study(studyId, username)
            ..name = 'Conditional Sampling Study'
            ..description = 'This is a study for testing and debugging Conditional Sampling'
            ..dataEndPoint = getDataEndpoint(DataEndPointTypes.FILE)
            ..addTriggerTask(
                ConditionalSamplingEventTrigger(
                    measureType: MeasureType(NameSpace.CARP, ContextSamplingPackage.LOCATION),
                    resumeCondition: (datum) {
                      return true;
                    },
                    pauseCondition: (datum) {
                      return true;
                    }),
                AutomaticTask()
                  ..measures = SamplingSchema.debug().getMeasureList(
                    namespace: NameSpace.CARP,
                    types: [
                      ContextSamplingPackage.WEATHER,
                      //ContextSamplingPackage.AIR_QUALITY,
                    ],
                  ))
            ..addTriggerTask(
                PeriodicTrigger(period: 1 * 20 * 1000, duration: 2 * 1000),
                //ImmediateTrigger(),
                AutomaticTask()
                  ..measures = SamplingSchema.debug().getMeasureList(
                    namespace: NameSpace.CARP,
                    types: [
                      ContextSamplingPackage.LOCATION,
                      //ContextSamplingPackage.GEOLOCATION,
                    ],
                  ))
```

bardram, 2020-05-20:
I've released version `0.7.1` of CAMS which handles a null pointer exception in this trigger.

closed 2020-05-20 by bardram