carp-dk/carp.sensing-flutter #65: Missing support for CARPDataEndPoint with token

MadsVSChristensen, 2020-05-28:
CAMS has support for authenticating with a token to CARP. However, CARPDataEndPoint needs a password, so these to methods do not compliment each other. 

Proposed solution:
Add a Token variable to CARPDataEndPoint, and not assert password & username != null.

bardram, 2020-05-31:
CAMS already have support for this. You don't need to specify a username, password, etc. in a `CARPDataEndPoint` if you already have created a `CarpService` configured with a `CarpApp` with prober credential (username, pw, client secrets). If a `CarpService` is already configured, the your CARPDataEndPoint may look like this:

```
CarpDataEndPoint(
          CarpUploadMethod.DATA_POINT,
          name: 'CANS Testing Server',
          //uri: bloc.uri,
          //clientId: bloc.clientID,
          //clientSecret: bloc.clientSecret,
          //email: bloc.username,
          //password: bloc.password
        )
```

Note that the uri, client id, etc. are commented out for pedagogical reasons ;-) No need to provide this. The only thing needed to specify is the upload method.

bardram, 2020-05-31:
The above test code is from the CAMS testing app

* https://github.com/cph-cachet/carp.sensing-flutter/tree/master/carp_mobile_sensing_test_app

where you in the bloc can see that I create a `CarpService` before configuring sensing in the init method:

```
void init() async {
    // set global debug level
    globalDebugLevel = DebugLevel.DEBUG;

    _study ??= await manager.getStudy(testStudyId);

    debug('Study : $study');

    // creating a CarpApp, configuring the CarpService, and authenticate the user
    CarpApp app = new CarpApp(
        study: study,
        name: "Test",
        uri: Uri.parse(uri),
        oauth: OAuthEndPoint(clientID: clientID, clientSecret: clientSecret));

    debug('CarpApp : $app');

    CarpService.configure(app);
    CarpUser user = await CarpService.instance.authenticate(username: username, password: password);
    debug('Signed in user - $user');

    await sensing.init();
  }
```

closed 2020-05-31 by bardram