carp-dk/carp.sensing-flutter #66: StudyID for the CARP Web Services can only be an integer

barnabashomola, 2020-06-11:
`[ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: 'package:carp_webservices/carp_service/carp_datapoint.dart': Failed assertion: line 55 pos 12: 'int.tryParse(studyId) != null': Study ID (ea7d63f7-3c42-43e8-ba4f-2c02ae63a272) for the CARP Web Services can only be an integer.`

The Study ID can only be integer, although my Study ID is a string (`"ea7d63f7-3c42-43e8-ba4f-2c02ae63a272"`), therefore I can't upload `CARPDataPoint` because the `CARPDataPoint.fromDatum` constructor cannot be called. 

Is it by design and  my Study ID should be changed to integer or should CAMS support string Study ID's as well?

bardram, 2020-06-18:
This problem goes a while back to the old CARP web services. See this issue [#16](https://github.com/cph-cachet/carp.webservices/issues/16).

bardram, 2020-06-18:
Should be fixed with release `0.4.3` - see

* https://pub.dev/packages/carp_webservices#-changelog-tab-

closed 2020-06-18 by bardram