carp-dk/carp.sensing-flutter #71: CARP Health Package: Argument type error - updated "Health" package needs to be supported.

MadsVSChristensen, 2020-08-07:
Error:
```
../../flutter/.pub-cache/hosted/pub.dartlang.org/carp_health_package-0.8.0/lib/health_domain.dart:109:23: Error: The argument type 'DateTime' can't be assigned to the parameter type 'int'.
 - 'DateTime' is from 'dart:core'.
      healthDataPoint.dateFrom,
                      ^
../../flutter/.pub-cache/hosted/pub.dartlang.org/carp_health_package-0.8.0/lib/health_domain.dart:110:23: Error: The argument type 'DateTime' can't be assigned to the parameter type 'int'.
 - 'DateTime' is from 'dart:core'.
      healthDataPoint.dateTo,
                      ^
```

Proposed solution:
Change HealthDatum constructor arguments 'dateFrom' & 'dateTo' to DateTime's. In "Health" these fields are now private and the getter for these returns DateTime's. 

```
  HealthDatum(this.value, this.unit, int dateFrom, int dateTo, this.dataType, this.platform) : super() {
    this.dateFrom = DateTime.fromMillisecondsSinceEpoch(dateFrom);
    this.dateTo = DateTime.fromMillisecondsSinceEpoch(dateTo);
  }
```
to 
```
  HealthDatum(this.value, this.unit, DateTime dateFrom, DateTime dateTo, this.dataType, this.platform) : super();

```

thomasnilsson, 2020-08-07:
Will do, also creating a UUID should be done in the CARP Health Package and not inside the Health plugin such that DASES HealthDataPoint objects can be created without specifying a UUID.

thomasnilsson, 2020-08-12:
@MadsVSChristensen Does it work for you now?

MadsVSChristensen, 2020-08-13:
Yes, as of version 1.0.1 it works with health 2.0.2

closed 2020-08-13 by MadsVSChristensen