carp-dk/carp.sensing-flutter #80: RecurrentScheduledTrigger not triggered if app is not running - New use case

barnabashomola, 2020-08-19:
[RecurrentScheduledTrigger](https://pub.dev/documentation/carp_mobile_sensing/latest/domain/RecurrentScheduledTrigger-class.html) is used to issue for e.g. weekly questionnaires using the `carp_survey_package`. When the app is not running the questionnaires are not popping up, even though they should.

A possible new feature enhancement could be a trigger type which uses the phone's storage (Shared Preferences or something...) to store the trigger information so it's not only triggered when the app is actively running but also next time the app is opened (maybe even pushes a system notification to open the app).

bardram, 2020-08-27:
Notes on use case based on discussion 2020-08-27:

* if the app is not running, a scheduled trigger should still trigger
* e.g. if scheduled for Sunday at 12, but the app open at Monday it should still trigger
* but there is an expire date for this
* e.g. if we reach the next Sunday, the first one is invalid

bardram, 2020-08-31:
Released as part of version `0.8.8`.

closed 2020-08-31 by bardram