carp-dk/carp.sensing-flutter #92: pubspec.yaml of example app and code example in and carp_mobile_sensing package page

sc00n, 2020-09-04:
When I download the example app and try to run it I get many errors. For me, a solution was to change the pubsec yaml to include carp_context_package: ^0.9.0 (instead of ^0.8.5). Probably other packages also need another version for optimal use.

The example in https://pub.dev/packages/carp_mobile_sensing is not up to date with current version of carp_mobile_sensing. Here is a possible update:

```
void example() async {
    // Create a study using a File Backend
    Study study = Study("1234", "user@dtu.dk",
        name: "An example study",
        dataEndPoint: FileDataEndPoint()
          ..bufferSize = 500 * 1000
          ..zip = true
          ..encrypt = false);

    // add sensor collection from accelerometer and gyroscope
    // careful - these sensors generate a lot of data!
    study.addTriggerTask(
        DelayedTrigger(delay: Duration(seconds: 1)), // delay sampling for one second
        Task(name: 'Sensor Task')
          ..addMeasure(PeriodicMeasure(
            MeasureType(NameSpace.CARP, SensorSamplingPackage.ACCELEROMETER),
            frequency: Duration(seconds: 10), // sample every 10 secs
            duration: Duration(milliseconds: 2), // for 2 ms
          ))
          ..addMeasure(PeriodicMeasure(
            MeasureType(NameSpace.CARP, SensorSamplingPackage.GYROSCOPE),
            frequency: Duration(seconds: 20), // sample every 20 secs
            duration: Duration(milliseconds: 2), // for 2 ms
          )));

    /*
    study.addTriggerTask(
        PeriodicTrigger(period: Duration(days: 1)), // trigger sampling once pr. day
        Task(name: 'Task collecting a list of all installed apps')
          ..addMeasure(Measure(MeasureType(NameSpace.CARP, AppsSamplingPackage.APPS))));*/

    // creating measure variable to be used later
    PeriodicMeasure lightMeasure = PeriodicMeasure(
      MeasureType(NameSpace.CARP, SensorSamplingPackage.LIGHT),
      name: "Ambient Light",
      frequency: Duration(seconds: 11),
      duration: Duration(milliseconds: 700),
    );
    study.addTriggerTask(ImmediateTrigger(), Task(name: 'Light')..addMeasure(lightMeasure));

    // Create a Study Controller that can manage this study, initialize it, and start it.
    StudyController controller = StudyController(study);

    // await initialization before starting
    await controller.initialize();
    controller.start();

    // listening on all data events from the study
    controller.events.forEach(print);

    // listen on only CARP events
    controller.events.where((datum) => datum.format.namepace == NameSpace.CARP).forEach(print);

    // listen on LIGHT events only
    controller.events.where((datum) => datum.format.name == SensorSamplingPackage.LIGHT).forEach(print);

    // map events to JSON and then print
    controller.events.map((datum) => datum.toJson()).forEach(print);

    // listening on a specific probe registered in the ProbeRegistry
    // this is equivalent to the statement above
    ProbeRegistry.probes[SensorSamplingPackage.LIGHT].events.forEach(print);

    // subscribe to events
    StreamSubscription<Datum> subscription = controller.events.listen((Datum datum) {
      // do something w. the datum, e.g. print the json
      print(JsonEncoder.withIndent(' ').convert(datum));
    });

    // sampling can be paused and resumed
    controller.pause();
    controller.resume();

    // pause / resume specific probe(s)
    ProbeRegistry.lookup(SensorSamplingPackage.ACCELEROMETER).pause();
    ProbeRegistry.lookup(SensorSamplingPackage.ACCELEROMETER).resume();

    // adapt measures on the go - calling hasChanged() force a restart of
    // the probe, which will load the new measure
    lightMeasure
      ..frequency = Duration(seconds:12)
      ..duration = Duration(milliseconds:500)
      ..hasChanged();

    // disabling a measure will pause the probe
    lightMeasure
      ..enabled = false
      ..hasChanged();

    // once the sampling has to stop, e.g. in a Flutter dispose() methods, call stop.
    // note that once a sampling has stopped, it cannot be restarted.
    controller.stop();
    subscription.cancel();
  }
```

This has however the following commented out:
```
 /*
    study.addTriggerTask(
        PeriodicTrigger(period: Duration(days: 1)), // trigger sampling once pr. day
        Task(name: 'Task collecting a list of all installed apps')
          ..addMeasure(Measure(MeasureType(NameSpace.CARP, AppsSamplingPackage.APPS))));*/
```
As I don't know how to change that to be suitable for the most recent version. This is the problem:
`Undefined name 'AppsSamplingPackage'.`

Thank you for the package, I'm looking forward to explore it further in the coming weeks!

Best,

Merijn

bardram, 2020-09-16:
Thanks for pointing this out @sc00n -- will update the documentation. You're right - it doesn't match the current released version.

bardram, 2020-09-16:
Just released v. 0.9.0 in which the documentation should be updated.

closed 2020-09-16 by bardram