carp-dk/carp.sensing-flutter #97: passive sensing problem

sc00n, 2020-09-24:
I tried to use the package for passive sensing, but I can't get it to work.

I applied the best practice tips (https://github.com/cph-cachet/carp.sensing-flutter/wiki/5.-Best-Practice).

To test the package I simply created a new flutter project and added the following to _MyHomePageState

```
  @override
  void initState() {
    // TODO: implement initState
    super.initState();
    initCarp();
  }
  
  
  void initCarp() async {
    // Create a study using a local file to store data
    Study study = Study("2", 'user@cachet.dk',
        name: 'A study collecting ..',
        dataEndPoint: FileDataEndPoint()
          ..bufferSize = 500 * 1000
          ..zip = true
          ..encrypt = false);
    
    // but delay the sampling by 10 seconds
    study.addTriggerTask(
        ImmediateTrigger(),
        AutomaticTask(name: 'Sensor Task')
          ..addMeasure(PeriodicMeasure(MeasureType(NameSpace.CARP, SensorSamplingPackage.LIGHT),
              frequency: const Duration(seconds: 10), duration: const Duration(milliseconds: 100))));

    // Create a Study Controller that can manage this study.
    StudyController controller = StudyController(study);

    // await initialization before starting/resuming
    await controller.initialize();
    controller.resume();

    // listening and print all data events from the study
    controller.events.forEach(print);
  }

```

I also adapted the AndroidManifest as mentioned in the Readme.

When I read the generated files (e.g. carp/data/2/carp-data-2020-09-24-17-35-33-471360.json) I see that the app clearly logs every 10 second when I am in the app. **When the app is open but the screen is closed I barely get any log and when the app is closed I get no logs at all.** 

I tested with Android emulator (Pixel 2, API 28) and my phone (Samsung Galaxy A40, Android 9).

Am I doing something wrong/can I change something, or is passive sensing just an illusion with current Android battery restrictions.

Thank you for your help!

Merijn

bardram, 2020-10-09:
Hi @sc00n - we've just release a version `0.9.3`. From version `0.9.3` the CAMS [`carp_context_package`](https://pub.dev/packages/carp_context_package) make use of the [`carp_background_location`](https://pub.dev/packages/carp_background_location). If you include the context package also, and follow the instruction on how to modify the Android `Manifest.xml` file, this will help keep the app running in the background.

But you're right - keeping passive sensing running on Android is getting more and more restricted. We will continue to look into this.

bardram, 2020-10-10:
> When the app is open but the screen is closed I barely get any log 

Regarding the background sampling - if you look at Figure 6 in the CAMS paper

 * https://arxiv.org/abs/2006.11904
 * https://arxiv.org/pdf/2006.11904.pdf

you can see that there is close to 100% sampling coverage both on Android and iOS. Also when the app is in the background. However, these test were done on a Samsung Galaxy A3, Android OS level 8.0.

I'm planning to rerun these test on Android 9.0 soon to investigate.

> and when the app is closed I get no logs at all.

No - the app has to run for sampling to run. CAMS is linked into the app (this is the whole idea of CAMS). If you want background sampling independent of your app, you should look for other (native) approaches, like [AWARE](https://awareframework.com) or [mCerebrum](https://md2k.org/software-how-to/software-smartphone.html).

bardram, 2020-10-13:
I've now looked into this. It appear that the light sensor is "killed" on Android 9. 

Here are the same test as in Figure 6 in the paper, done on 

 * **SM-A320FL** : Samsung Galaxy A3, OS level 8.0, API 26
 * **SM-G970F** : Samsung Galaxy A3, OS level 10.0, API 29

[image]

[image]

bardram, 2020-10-13:
Also found this these [Best Practices for Accessing and Using Sensors](https://developer.android.com/guide/topics/sensors/sensors_overview#sensors-practices) in Android 9+.

koenniem, 2020-10-14:
The docs also say 
> Given these restrictions, it's best to detect sensor events either when your app is in the foreground or as part of a foreground service.

So what if the entire sensing class would run on a foreground service? That way, the sensing would stay active even if the app were closed and the sensors would get a higher priority. I wonder if and how this impacts battery life though.

bardram, 2020-10-28:
I've made this blog post

https://carp.cachet.dk/sampling-coverage-in-mobile-sensing-on-android-8-vs-10/

giving a status on this. Still testing it on iOS, though. 

Will also consider to add foreground service to CAMS core later -- issue #101

bardram, 2020-11-02:
Finally - after release of version `0.9.5` - I managed to run the coverage test on iOS. With good results, as the graph below shows.

[image]

Doing a blog post on this too....

closed 2020-11-09 by bardram