carp-dk/carp.sensing-flutter #98: stupid plugin , this error from 3 months ago and still as it is ..

imeDevelopers, 2020-09-29:
```
[ +301 ms] I/flutter (23953): ══╡ EXCEPTION CAUGHT BY SERVICES LIBRARY
╞══════════════════════════════════════════════════════════
[  +62 ms] I/flutter (23953): The following MissingPluginException was thrown while activating
platform stream on channel
[   +6 ms] I/flutter (23953): screenStateEvents:
[  +67 ms] I/flutter (23953): MissingPluginException(No implementation found for method listen  
on channel screenStateEvents)
[  +60 ms] I/flutter (23953): 
[  +52 ms] I/flutter (23953): When the exception was thrown, this was the stack:
[  +11 ms] I/flutter (23953): #0      MethodChannel._invokeMethod
(package:flutter/src/services/platform_channel.dart:157:7)
[  +60 ms] I/flutter (23953): <asynchronous suspension>
[  +52 ms] I/flutter (23953): #1      MethodChannel.invokeMethod
(package:flutter/src/services/platform_channel.dart:332:12)
[  +15 ms] I/flutter (23953): #2      EventChannel.receiveBroadcastStream.<anonymous closure>
(package:flutter/src/services/platform_channel.dart:540:29)
[  +53 ms] I/flutter (23953): #3      EventChannel.receiveBroadcastStream.<anonymous closure>
(package:flutter/src/services/platform_channel.dart:526:64)
```

bardram, 2020-10-02:
@imeDevelopers - if you're going to report issues, then please (1) be polite, (2) constructive, and (3) provide details on how you experience the problem. I'm going to change your title.

bardram, 2020-10-02:
This problem is cause by an issue in the Screen State plugin -- see https://github.com/cph-cachet/flutter-plugins/issues/188

bardram, 2020-11-11:
Fixed by the upgrade of the other issue #118.

closed 2020-11-11 by bardram