carp-dk/carp.sensing-flutter #101: Enable foreground service on Android

bardram, 2020-10-07:
Just like in the mCardia app, we could consider to have CAMS to notify a foreground service and thereby keep the app in the foreground.

* Android [Service](https://developer.android.com/reference/android/app/Service) API.
* movisenslibrary [Service](https://github.com/cph-cachet/movisenslibrary/blob/master/src/main/java/de/kn/uni/smartact/movisenslibrary/bluetooth/MovisensService.java) class.
* [Movisens Bluetooth Example Project](https://github.com/movisens/MovisensBluetoothSample)

sc00n, 2020-10-08:
I think this is a good idea. If I remember correctly, the EARS platform (https://www.ksanahealth.com/ears-research-platform) also uses this. Coincidently I am currently also experimenting with Android foreground services, to keep sensors running while the app is closed, and until now it seems to work fine. I started from this package: https://pub.dev/packages/foreground_service.

bardram, 2020-11-25:
This can be done with our new `[flutter_foreground_service](https://pub.dev/packages/flutter_foreground_service)` package.

bardram, 2022-06-21:
This should not be part of CAMS but part of the app itself.

closed 2022-06-21 by bardram