carp-dk/carp.sensing-flutter #104: Missing permission for Android Activity Recognition on Android 10

koenniem, 2020-10-20:
**The problem**
Activity recognition does not work on Android 10: 
`requestActivityUpdates: Failed to enable Activity updates: 10: SecurityException: Activity detection usage requires the ACTIVITY_RECOGNITION permission`

**How to reproduce**
Run the example app on Android 10 with the `ContextSamplingPackage.ACTIVITY` probe activated.

**In more detail**
According to the [docs](https://developer.android.com/about/versions/10/privacy/changes#physical-activity-recognition):

> If your app targets Android 9 (API level 28) or lower, the system auto-grants the `android.permission.ACTIVITY_RECOGNITION` permission to your app, as needed, if your app satisfies each of the following conditions:
> 
> - The manifest file includes the `com.google.android.gms.permission.ACTIVITY_RECOGNITION` permission.
> - The manifest file doesn't include the `android.permission.ACTIVITY_RECOGNITION` permission.
> 
> If the system-auto grants the `android.permission.ACTIVITY_RECOGNITION` permission, your app retains the permission after you update your app to target Android 10. However, the user can revoke this permission at any time in system settings.

So, while the `com.google.android.gms.permission.ACTIVITY_RECOGNITION` is present in in the Android Manifest, it doesn't request access on Android 10. Simply adding the permission is not possible because that would break the automatic access on Android 9.0 or lower. 

**Possible solution**
Ask for `android.permission.ACTIVITY_RECOGNITION` permission from the user without putting it in the manifest. Or put both in the manifest and ask for the appropriate one at runtime.

bardram, 2020-10-21:
Yes - this is seems to be related to the `activity_recognition_flutter` plugin -- see [issue #215](https://github.com/cph-cachet/flutter-plugins/issues/215).

One option is to compile for a lower SDK, and change the `compileSdkVersion` to `28` is the `build.graddle` file.

bardram, 2020-10-25:
By solving [Issue #215](https://github.com/cph-cachet/flutter-plugins/issues/215) i the `activity_recognition_flutter` plugin this issue is now also solved. There is a new version of the [demo app](https://github.com/cph-cachet/carp.sensing-flutter/tree/master/apps/carp_mobile_sensing_app) which is complied to Android 10 / SDK 29 which enables activity recognition.

commit 5c43096d (2020-10-25): Fixed #104
commit e0f07efc (2020-10-25): Merge pull request #109 from cph-cachet/develop

Fixed #104
closed 2020-10-25 by bardram