carp-dk/carp.sensing-flutter #113: Multiple RecurrentScheduledTriggers doesnt trigger the ConditionalEventTriggers

MadsVSChristensen, 2020-10-29:
A sensins as such WORKS:
```
// Conditional
..addTriggerTask(
<some conditional trigger>,
<some task>)
..addTriggerTask(
<some conditional trigger>,
<some task>)

// Recurrent
..addTriggerTask(
<some recurrent trigger>,
<some task>)
```
However a sensing with multiple ReccurentScheduledTriggers does NOT WORK:
```
// Conditional
..addTriggerTask(
<some conditional trigger>,
<some task>)
..addTriggerTask(
<some conditional trigger>,
<some task>)

// Recurrent
..addTriggerTask(
<some recurrent trigger>,
<some task>)
..addTriggerTask(
<some recurrent trigger>,
<some task>)
```
It doesnt trigger the conditional statement.

bardram, 2020-11-02:
I'm not quite sure I understand this. Is the problem only that multiple ReccurentScheduledTriggers in the same Study don't work? What is the role of the ConditionalSamplingEventTrigger?

MadsVSChristensen, 2020-11-05:
The issue is that when i have multiple `RecurrentScheduledTrigger`s then they do NOT check the `ConditionalSamplingEventTrigger`s. The `resumeCondition` isn't even triggered.

However it DOES work if there is only ONE `RecurrentScheduledTrigger`.

(The resume conditions can be whatever - just return true if you want. But the key is that they do not even get checked.)

Code example: 
```
DateTime now = DateTime.now();
_study = Study(studyID, sp.getString(CARPBackend.SP_USER_ID))
            ..name = studyID
            ..deploymentId = sp.getString(CARPBackend.SP_DEPLOYMENT_ID)
            ..description = "This is a study for iPDM-GO #${sp.getString(CARPBackend.SP_STUDY_ID)}"
            ..dataEndPoint = getDataEndpoint(DataEndPointTypes.CARP)

            // Conditional surveys based on other surveys 
            // THESE DO NOT GET CHECKED WHEN I HAVE MULTIPLE RecurrentScheduledTriggers!
            ..addTriggerTask(
                ConditionalSamplingEventTrigger(
                    measureType: MeasureType(NameSpace.CARP, SurveySamplingPackage.SURVEY),
                    resumeCondition: sensingBloc.resumeConditionD1),
                AppTask(name: 'Food behavior', onResume: sensingBloc.addTaskWithSurvey)
                  ..measures.add(RPTaskMeasure(
                    MeasureType(NameSpace.CARP, SurveySamplingPackage.SURVEY),
                    enabled: true,
                    surveyTask: surveyTaskD1,
                  )))
            ..addTriggerTask(
                ConditionalSamplingEventTrigger(
                    measureType: MeasureType(NameSpace.CARP, SurveySamplingPackage.SURVEY),
                    resumeCondition: sensingBloc.resumeConditionD2),
                AppTask(name: 'Sleep quality and patterns (PSQI)', onResume: sensingBloc.addTaskWithSurvey)
                  ..measures.add(RPTaskMeasure(
                    MeasureType(NameSpace.CARP, SurveySamplingPackage.SURVEY),
                    enabled: true,
                    surveyTask: surveyTaskD2,
                  )))
            ..addTriggerTask(
                ConditionalSamplingEventTrigger(
                    measureType: MeasureType(NameSpace.CARP, SurveySamplingPackage.SURVEY),
                    resumeCondition: sensingBloc.resumeConditionD3),
                AppTask(name: 'Depression and anxiety (HADS)', onResume: sensingBloc.addTaskWithSurvey)
                  ..measures.add(RPTaskMeasure(
                    MeasureType(NameSpace.CARP, SurveySamplingPackage.SURVEY),
                    enabled: true,
                    surveyTask: surveyTaskD3,
                  )))
            ..addTriggerTask(
                ConditionalSamplingEventTrigger(
                    measureType: MeasureType(NameSpace.CARP, SurveySamplingPackage.SURVEY),
                    resumeCondition: sensingBloc.resumeConditionD4),
                AppTask(name: 'Medication behavior (MA)', onResume: sensingBloc.addTaskWithSurvey)
                  ..measures.add(RPTaskMeasure(
                    MeasureType(NameSpace.CARP, SurveySamplingPackage.SURVEY),
                    enabled: true,
                    surveyTask: surveyTaskD4,
                  )))

            //
            ..addTriggerTask(
                RecurrentScheduledTrigger(
                  triggerId: 'Well-being trigger',
                  type: RecurrentType.weekly,
                  dayOfWeek: now.weekday,
                  time: Time(hour: now.hour, minute: now.minute + 1),
                  remember: true,
                ),
                AppTask(name: 'Well-being', onResume: sensingBloc.addTaskWithSurvey)
                  ..measures.add(RPTaskMeasure(
                    MeasureType(NameSpace.CARP, SurveySamplingPackage.SURVEY),
                    enabled: true,
                    surveyTask: wellbeing,
                  )))
            ..addTriggerTask(
                RecurrentScheduledTrigger(
                  triggerId: 'Emotional Distress trigger',
                  type: RecurrentType.weekly,
                  dayOfWeek: now.weekday,
                  time: Time(hour: now.hour, minute: now.minute + 1),
                  remember: true,
                ),
                AppTask(name: 'Emotional Distress', onResume: sensingBloc.addTaskWithSurvey)
                  ..measures.add(RPTaskMeasure(
                    MeasureType(NameSpace.CARP, SurveySamplingPackage.SURVEY),
                    enabled: true,
                    surveyTask: emotionalDistress,
                  )));
```

bardram, 2020-11-11:
I've been looking into this, and I think the problem is that you don't specify the `pauseCondition` for the `ConditionalSamplingEventTrigger`. When the pause condition is not specified, the trigger resumes **once** and is never paused. And you cannot resumed a paused trigger, so you probably get a warning.

I tried to specify the following `pauseCondition`;

```
pauseCondition: (Datum datum) => true,
```

which worked in my case. But uncertain if it works in yours.

bardram, 2020-11-25:
I did, however, find another bug, which might be related to this one. This bug is fixed as part of version `0.10.0` which will be released soon.

closed 2020-11-29 by bardram