carp-dk/carp.sensing-flutter #115: Context package necessary for background sensing (small enhancement)

sc00n, 2020-10-29:
For both iPhone and Android the context package seems to be necessary to do any kind of background sensing. This is kind of stated for Android, but not for iOS. In iOS it only states that you need to use this package if you are interested in location.

I would add a general comment in the beginning such as:

> To keep the app running in the background, the carp_context_package needs to be included. Please follow the instructions [here](https://pub.dev/packages/carp_context_package) on how to use this package.

bardram, 2020-11-02:
Where are you reading? Where should we add this information?

sc00n, 2020-11-02:
I meant on the best practices page: https://github.com/cph-cachet/carp.sensing-flutter/wiki/5.-Best-Practice. The information on the context page could be somewhere in the beginning, as it is OS independent. Or it can be in both iOS and Android section.

bardram, 2020-11-03:
ok - done.

closed 2020-11-03 by bardram