carp-dk/carp.sensing-flutter #123: Anonymise function in PrivacySchema gets called 4 times

koenniem, 2020-11-23:
carp_mobile_sensing v0.9.5

### Device / Emulator and OS
 - Device: Pixel 4
 - OS: Android API 30

### Describe the bug
When applying a PrivacySchema, anonymiser functions are called 4 times while the actual function gets applied 3 times. For example, `wifiNameAnoymizer(Datum datum)` is called  4 times while the `sha1` function is applied 3 times. Check this by going to [a website for sha1](http://www.sha1-online.com/) and hash "AndroidWifi" (Wi-Fi name in the emulator):
* Expected: 8996a5c820f919bad6fce65f2532aa430148eee9
* Actual: 10fee701d4a210776d5cfe561d08ad817935216d
* Expected x3: 10fee701d4a210776d5cfe561d08ad817935216d

This happens with other probes as well. The problem persisted when I tried to extend CARP with my own privacy schema.

### To Reproduce
Run the example app with a privacy schema, e.g. PrivacySchema.DEFAULT. 

### Expected behavior
The anonymiser function is supposed to be called only one time.

### Actual behavior
The anonymiser function was called 4 times but applied on the 3rd. 

### Stack trace (of bluetooth anonymiser)
```dart
blueetothNameAnoymizer (connectivity_privacy.dart:6)
TransformerSchema.transform (transformers.dart:77)
new StudyController.<anonymous closure> (study_controller.dart:84)
_MapStream._handleData (stream_pipe.dart:219)
_ForwardingStreamSubscription._handleData (stream_pipe.dart:157)
_rootRunUnary (zone.dart:1198)
_CustomZone.runUnary (zone.dart:1100)
_CustomZone.runUnaryGuarded (zone.dart:1005)
_BufferingStreamSubscription._sendData (stream_impl.dart:357)
_BufferingStreamSubscription._add (stream_impl.dart:285)
_SyncBroadcastStreamController._sendData.<anonymous closure> (broadcast_stream_controller.dart:393)
_BroadcastStreamController._forEachListener (broadcast_stream_controller.dart:330)
_SyncBroadcastStreamController._sendData (broadcast_stream_controller.dart:392)
_BroadcastStreamController.add (broadcast_stream_controller.dart:250)
_rootRunUnary (zone.dart:1198)
_CustomZone.runUnary (zone.dart:1100)
_CustomZone.runUnaryGuarded (zone.dart:1005)
_BufferingStreamSubscription._sendData (stream_impl.dart:357)
_BufferingStreamSubscription._add (stream_impl.dart:285)
_SyncBroadcastStreamController._sendData (broadcast_stream_controller.dart:385)
_BroadcastStreamController.add (broadcast_stream_controller.dart:250)
_rootRunUnary (zone.dart:1198)
_CustomZone.runUnary (zone.dart:1100)
_CustomZone.runUnaryGuarded (zone.dart:1005)
_BufferingStreamSubscription._sendData (stream_impl.dart:357)
_BufferingStreamSubscription._add (stream_impl.dart:285)
_SyncBroadcastStreamController._sendData (broadcast_stream_controller.dart:385)
_BroadcastStreamController.add (broadcast_stream_controller.dart:250)
_rootRunUnary (zone.dart:1198)
_CustomZone.runUnary (zone.dart:1100)
_CustomZone.runUnaryGuarded (zone.dart:1005)
_BufferingStreamSubscription._sendData (stream_impl.dart:357)
_DelayedData.perform (stream_impl.dart:611)
_StreamImplEvents.handleNext (stream_impl.dart:730)
_PendingEvents.schedule.<anonymous closure> (stream_impl.dart:687)
_rootRun (zone.dart:1182)
_CustomZone.run (zone.dart:1093)
_CustomZone.runGuarded (zone.dart:997)
_CustomZone.bindCallbackGuarded.<anonymous closure> (zone.dart:1037)
_rootRun (zone.dart:1190)
_CustomZone.run (zone.dart:1093)
_CustomZone.runGuarded (zone.dart:997)
_CustomZone.bindCallbackGuarded.<anonymous closure> (zone.dart:1037)
_microtaskLoop (schedule_microtask.dart:41)
_startMicrotaskLoop (schedule_microtask.dart:50)
```

bardram, 2020-12-09:
I've been looking into this and here are my notes (mostly for myself for later reference. 

The bug is real and it is related to listening to the `events` stream. It seems like the anonymize functions gets called for every time a subscriber listens to the `events` stream. This is strange, and after hours of debugging, I can't spot why. 

In order to test a simple setup, I'm using this Dart code.

```
void main() {
  var counterStream = Stream<int>.periodic(Duration(seconds: 1), (x) => x)
      .take(15)
      .asBroadcastStream()
      .map((i) => 2 * i);

  final subscription1 = counterStream.listen(
    (data) => print('Data-1: $data'),
  );

  final subscription2 = counterStream.listen(
    (data) => print('Data-2: $data'),
  );
}
```

bardram, 2020-12-09:
Think I got it now! The problem is the order of the `asBroadcastStream()` modifier......

The code above should read like this:

```
void main() {
  var counterStream = Stream<int>.periodic(Duration(seconds: 1), (x) => x)
      .take(15)
      .map((i) => 2 * i)
      .asBroadcastStream();

  final subscription1 = counterStream.listen(
    (data) => print('Data-1: $data'),
  );

  final subscription2 = counterStream.listen(
    (data) => print('Data-2: $data'),
  );
}
```

bardram, 2020-12-09:
Bug fix will be released as part of version`0.10.1`

closed 2020-12-09 by bardram