carp-dk/carp.sensing-flutter #134: Accelerometer frequency and duration

koenniem, 2020-12-02:
Prior to `carp_mobile_sensing: 0.10:0` the accelerometer worked fine like this

```dart
      MapEntry(
          SensorSamplingPackage.ACCELEROMETER,
          PeriodicMeasure(MeasureType(NameSpace.CARP, SensorSamplingPackage.ACCELEROMETER),
              name: 'Accelerometer',
              enabled: true,
              frequency: const Duration(milliseconds: 1000),
              duration: const Duration(milliseconds: 10))),
```
This would give me 1 measurement per second with 5 samples in the `MultiDatum`. 

However, after version 0:10.0 was released, this no longer works and instead it gives something like
```dart
MultiDatum - format: unknown.unknown, id: 6eccc2d0-34b5-11eb-a996-8ffab3efffeb, timestamp: 2020-12-02 15:45:41.756658Z, size: 0
````
Only when I increase the frequency and duration to something like 3 and 1 seconds respectively does it work properly again, now with 5 samples in the `MultiDatum` every 3 seconds. 

My questions:
* How do I increase the number of measurements to, say, every second?
* What does the duration actually do?

bardram, 2020-12-07:
Yes - when fixing #114, the number of event from the accelerometer (and gyroscope) was dramatically reduced since they were no longer buffered. 

Now, if you look into the default settings now in

 * https://github.com/cph-cachet/carp.sensing-flutter/blob/master/carp_mobile_sensing/lib/sampling_packages/sensors/sensor_package.dart

these are now;

```
PeriodicMeasure(MeasureType(NameSpace.CARP, ACCELEROMETER),
              name: 'Accelerometer',
              enabled: false,
              frequency: const Duration(seconds: 5),
              duration: const Duration(seconds: 1)))
```

i.e. 5 and 1 seconds, respectively.

bardram, 2020-12-07:
> What does the duration actually do?

bardram, 2020-12-07:
The `format: unknown.unknown` is, however, a bug.

bardram, 2020-12-07:
I have seriously been considering to **not** use a `PeriodicMeasure` but a "normal" measure, in which the accelerometer sampling is triggered by a `Trigger` instead. 

What is you view on this? What is the scenario / use case you want to use accelerometer data for?

koenniem, 2020-12-07:
> In a `PeriodicMeasure` it samples every frequency for the specified duration. Hence, in the example above, you would collect accelerometer events every 5 seconds for 1 second per sampling window.

I don't know why, but it seems to be working much better now. Before it took 5 samples per frequency now it's around 14 samples. I can imagine now that the subscription gets cancelled each time it is no longer feasible to have a low frequency, but I guess that's also not needed (normally).

> The `format: unknown.unknown` is, however, a bug.

That's good to hear. I got that before version `0.10.0` a few times as well but I suppose it was for the same reason.

> I have seriously been considering to not use a `PeriodicMeasure` but a "normal" measure, in which the accelerometer sampling is triggered by a `Trigger` instead.
> 
> What is you view on this? What is the scenario / use case you want to use accelerometer data for?

I'm not sure how it would be triggered if not by a `PeriodicTrigger`. That is, what do you get when using a normal `Trigger`?

Our "use case" is that we want to collect "adequate" accelerometer data which we can use to infer movement. This can be to estimate sleep or another measure for physical activity. So, it's not that we need very high-frequent measurements of the accelerometer per se, I just thought a 1 second sample per 5 seconds would not be enough, but I might be wrong. Also, I just thought it was odd it didn't work all of a sudden.

Now that I'm trying it again it seems to be working much better. Given that the samples per frequency are sampled uniformly, this is great for our purposes. Admittedly, specifying both a `duration` and `frequency` may not be very intuitive since it's more the ratio `duration`/`frequency` that matters than their absolute values.

bardram, 2020-12-09:
I've implemented support for both types of measures - continuously and periodic. See also the [wiki](https://github.com/cph-cachet/carp.sensing-flutter/wiki/A.-Measure-Types).

In version 0.10.1 the following measures has changed:

    * `periodic_accelerometer` and `periodic_gyroscope` has been added
    * the "old" `accelerometer` and `gyroscope` measures are made to run continously
    * This implies that they need to be started and stopped using a `Trigger` instead. 
    * This, however, open up for triggering the collection of accelerometer data as part of an `AppTask`. For example, sampling accelerometer data while a Parkinson's patient fill in a survey.

closed 2020-12-09 by bardram