carp-dk/carp.sensing-flutter #139: register failed, the sensor listeners size has exceeded the maximum limit 128

sc00n, 2020-12-16:
When I use the simple Flutter example app and add the light measure to my CARP study I get the following error (tested on Android phone and emulator):

<details>
<summary>Failed to open event stream</summary>
<br>

```
E/EventChannel#light.eventChannel( 8803): java.lang.IllegalStateException: register failed, the sensor listeners size has exceeded the maximum limit 128
E/EventChannel#light.eventChannel( 8803): 	at android.hardware.SystemSensorManager.registerListenerImpl(SystemSensorManager.java:197)
E/EventChannel#light.eventChannel( 8803): 	at android.hardware.SensorManager.registerListener(SensorManager.java:859)
E/EventChannel#light.eventChannel( 8803): 	at android.hardware.SensorManager.registerListener(SensorManager.java:766)
E/EventChannel#light.eventChannel( 8803): 	at dk.cachet.light.LightPlugin.onListen(LightPlugin.java:54)
E/EventChannel#light.eventChannel( 8803): 	at io.flutter.plugin.common.EventChannel$IncomingStreamRequestHandler.onListen(EventChannel.java:188)
E/EventChannel#light.eventChannel( 8803): 	at io.flutter.plugin.common.EventChannel$IncomingStreamRequestHandler.onMessage(EventChannel.java:167)
E/EventChannel#light.eventChannel( 8803): 	at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(DartMessenger.java:85)
E/EventChannel#light.eventChannel( 8803): 	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(FlutterJNI.java:692)
E/EventChannel#light.eventChannel( 8803): 	at android.os.MessageQueue.nativePollOnce(Native Method)
E/EventChannel#light.eventChannel( 8803): 	at android.os.MessageQueue.next(MessageQueue.java:326)
E/EventChannel#light.eventChannel( 8803): 	at android.os.Looper.loop(Looper.java:181)
E/EventChannel#light.eventChannel( 8803): 	at android.app.ActivityThread.main(ActivityThread.java:7063)
E/EventChannel#light.eventChannel( 8803): 	at java.lang.reflect.Method.invoke(Native Method)
E/EventChannel#light.eventChannel( 8803): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494)
E/EventChannel#light.eventChannel( 8803): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:965)

════════ Exception caught by services library ══════════════════════════════════════════════════════
The following PlatformException was thrown while activating platform stream on channel light.eventChannel:
PlatformException(error, register failed, the sensor listeners size has exceeded the maximum limit 128, null, null)
```

</details>

To reproduce you can add the following code: 

```
    PeriodicMeasure lightMeasure = PeriodicMeasure(
      MeasureType(NameSpace.CARP, SensorSamplingPackage.LIGHT),
      name: "Ambient Light",
      frequency: const Duration(seconds: 1),
      duration: const Duration(milliseconds: 100),
    );
```

After about 128 seconds you get the error. You can also increase the duration in the frequency parameter (e.g. by using SamplingSchema.common()), but in that case you have to wait longer to reproduce the error. I think I get the error because the stream is not correctly cancelled. 

When I was playing around myself with foreground services a while ago I had the same problem when I used your [light](https://pub.dev/packages/light) package. I solved it by adapting the [sensor package](https://pub.dev/packages/sensors) to include a light sensor. This package and code did not seem to have the cancelling problem.

<details>
<summary>Flutter doctor -v</summary>
<br>

```
[√] Flutter (Channel stable, 1.22.4, on Microsoft Windows [Version 10.0.19041.630], locale nl-BE)
    • Flutter version 1.22.4 at C:\src\flutter\flutter
    • Framework revision 1aafb3a8b9 (5 weeks ago), 2020-11-13 09:59:28 -0800
    • Engine revision 2c956a31c0
    • Dart version 2.10.4

[√] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
    • Android SDK at C:\Users\u0084978\AppData\Local\Android\sdk
    • Platform android-30, build-tools 29.0.2
    • Java binary at: C:\Program Files\Android\Android Studio2\jre\bin\java
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b01)
    • All Android licenses accepted.

[!] Android Studio (version 4.1.0)
    • Android Studio at C:\Program Files\Android\Android Studio2
    X Flutter plugin not installed; this adds Flutter specific functionality.
    X Dart plugin not installed; this adds Dart specific functionality.
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b01)

[√] Connected device (2 available)
    • SM A405FN (mobile)           • R58N11LA7QV   • android-arm64 • Android 9 (API 28)
    • AOSP on IA Emulator (mobile) • emulator-5554 • android-x86   • Android 9 (API 28) (emulator)
```

</details>

I use `carp_mobile_sensing: ^0.10.1`

koenniem, 2020-12-16:
My guess (but it's just a guess) is that the `Timer` expires before the subscription has started. In `BufferingPeriodicStreamProbe`:

``` dart
  Future<void> onResume() async {
    timer = Timer.periodic(frequency, (Timer t) {
      onSamplingStart();
      subscription = bufferingStream?.listen(onSamplingData,
          onError: onError, onDone: onDone);
      Timer(duration, () async {
        await subscription?.cancel();
        onSamplingEnd();
        await getDatum().then((datum) {
          if (datum != null) controller.add(datum);
        }).catchError(
            (error, stacktrace) => controller.addError(error, stacktrace));
      });
    });
  }
```` 
The `cancel()` method called upon `subscription` is null-safe and it is therefore possible `cancel()` is called before the `subscription` has finished (and thus has no effect). 

Before issue #114 the stream would have been paused but this would not cause the error.

sc00n, 2020-12-16:
I tested if the cancel was actually called by setting the oncancel method in the android code of the light package to

```
    @Override
    public void onCancel(Object arguments) {
        /// Finish listening to events
		System.out.println("Do the cancellation");
        sensorManager.unregisterListener(sensorEventListener);
		System.out.println("Done the cancellation");
    }
```
and it seems to be called every time.

sc00n, 2020-12-16:
Ok, I solved the bug. In the onlisten you put

`SensorEventListener sensorEventListener = createSensorEventListener(events);`
but this should be
`sensorEventListener = createSensorEventListener(events);`

Otherwise your class variable `sensorEventListener ` stays null.  

I will now try to do my very first pull request, as I was not a frequent Githubber before I started using Flutter :)

bardram, 2020-12-16:
Thanks. Has merged your PR now. Closing this ticket.

closed 2020-12-17 by bardram