carp-dk/carp.sensing-flutter #143: Health 1.2.0 depends on carp_mobile_sensing 0.10.0

koenniem, 2020-12-18:
As said in the title, there is a dependency issue between `carp_health_package: 1.2.0` and `carp_mobile_sensing: 0.11.0`:

```
Because carp_health_package 1.2.0 depends on carp_mobile_sensing ^0.10.0 and no versions of carp_health_package match >1.2.0 <2.0.0, carp_health_package ^1.2.0 requires carp_mobile_sensing ^0.10.0.
So, because flutter_app depends on both carp_mobile_sensing ^0.11.0 and carp_health_package ^1.2.0, version solving failed.
```

This will probably be fixed when upgrade `carp_health_package` to 3.0.0.

koenniem, 2020-12-18:
The same thing applies to `carp_communications_package` by the way.

bardram, 2020-12-21:
I know - had to go over all the packages -- should be released now.

closed 2020-12-21 by bardram