carp-dk/carp.sensing-flutter #172: Upgrade permission_handler to version 6

koenniem, 2021-03-08:
At the moment, all CARP packages depends on `permission_handler ^5.0.0` while the most recent version in 6. This version solves a critical bug in the way API 30 handles permission, see [here](https://github.com/Baseflow/flutter-permission-handler/wiki/Changes-in-6.0.0) and [here](https://github.com/Baseflow/flutter-geolocator/wiki/Breaking-changes-in-7.0.0).

bardram, 2021-03-15:
I have been looking into this, but this would require that CAMS is upgraded to handle null-safety -- which is on the todo list, but not yet.

bardram, 2021-06-21:
FYI - I'm working on CAMS null-safety. Will be released soon...

bardram, 2021-08-03:
CAMS released in null-safe version as version 0.30.x

closed 2021-08-03 by bardram