carp-dk/carp.sensing-flutter #183: [carp_communication_package 0.12.0] Using this package implies requesting restricted permissions

koenniem, 2021-04-25:
## Motivation
We wish to publish our app, which uses various CARP package, to the Google Play Store. However, initially the app was rejected because we tried to collect SMS texts and call logs which Google considers to be a [restricted group of permissions](https://support.google.com/googleplay/android-developer/answer/10208820?visit_id=637547856829628711-3104765488&rd=1&hl=en) for which you need a valid reason to access those. Unfortunately, research is not in the list of exceptions. Hence, we would like to remove collection of SMS texts and call logs while still keeping collection of calendar information (all in the `carp_communication_package`).

## The problem
Upon adding `carp_communication_package 12.0.0` (or any later or earlier version for that matter), various restricted permissions such as `android.permission.READ_SMS` and `android.permissions.REAL_CALL_LOG` are automatically added because they reside in the [`call_log`](https://pub.dev/packages/call_log) and [`sms`](https://pub.dev/packages/sms) plugin that the package uses. As a consequence, the app is automatically rejected from the Play Store. Unfortunately, this problem persists even when removing call log and SMS texts from the sensing list, since these plugins are loaded immediately upon registering `carp_communication_package`, regardless of whether the plugins will be used or not.

## Resolution
I'm not sure whether there could be any solution for this problem since it would almost mean disassembling the package itself. Perhaps there exists some code which prevents the permission from being requested or the plugin from being loaded? At least the documentation could use a warning for other users puzzling why problems occurs when trying to get an app published to the Play Store.

koenniem, 2021-04-25:
Actually I have just found the solution. The problem indeed occurs when merging [manifest files](https://developer.android.com/studio/build/manifest-merge.html). This means the illegal permissions can be omitted by specifying that they should not be merged from a lower level manifest. I suggest adding this information to the documentation of this package.

Concretely, adding the following lines to the AndroidManifest prevents the restricted permissions from being added:
``` XML
<uses-permission android:name="android.permission.READ_CALL_LOG" tools:node="remove"/>
<uses-permission android:name="android.permission.READ_SMS" tools:node="remove"/>
<uses-permission android:name="android.permission.SEND_SMS" tools:node="remove"/>
<uses-permission android:name="android.permission.RECEIVE_SMS" tools:node="remove"/>
```

bardram, 2021-05-07:
Thanks for the input. 

But exactly - in which AndroidManifest should this be added? In the

 * `carp_communication_package`
 * `carp_mobile_sensing`
 * the app you're building

?

bardram, 2021-06-07:
So - as I understand the above: 

There is no need to change the `carp_communication_package` as such. We should just add the explanation that it is possible to "remove" the permissions if not used in the app.

Right?

koenniem, 2021-06-07:
That's correct. As long as developers are aware of this issue, it shouldn't raise any further problems.

bardram, 2021-10-14:
Added the the "Tips'n Tricks" page 

 * https://github.com/cph-cachet/carp.sensing-flutter/wiki/5.-Best-Practice#android-manifest-file-merge