carp-dk/carp.sensing-flutter #191: lib/src/sensing/credentials.dart Not Found

LucyCat-byte, 2021-06-02:
I am new to this package and just try to build the carp_mobile_sensing_app on my Android Studio. It seems this file is needed as a demo to pass authentication, but it is currently not found in the directory. 

Should I do any modification to set up the app? Thank you for any help!

bardram, 2021-06-07:
Hi @LucyCat-byte - 

This file is only used when you want to upload data to our CARP backend (https://cans.cachet.dk). And since this server is only used by us, you don't need the credentials files.

However, I can see it's a bad idea to rely on a file that isn't in the github repro. I will refactor this, so that you will not get a compile error when cloning the repro.

Moreover, when you want to run the app locally, make sure to confiture it to run locally. This is done in the `app.dart` in line `16` like this:

```
// initialize the bloc, setting the deployment mode:
    //  * LOCAL
    //  * CARP_STAGGING
    //  * CARP_PRODUCTION
    await bloc.initialize(DeploymentMode.LOCAL);
```

Sorry for this confusion - it's because we're using the same app for testing both locally and remote deployments....

bardram, 2021-06-07:
@LucyCat-byte - I've updated the CAMS demo app and removed the credentials file - have a look now.

closed 2021-06-07 by bardram