carp-dk/carp.sensing-flutter #198: Unknown key when creating deployment using CarpDeploymentService

carlos-bailon, 2021-06-09:
When I try to create a study deployment using [CarpDeploymentService](https://github.com/cph-cachet/carp.sensing-flutter/blob/master/backends/carp_webservices/lib/carp_services/deployment_service.dart) from [carp_webservices](https://github.com/cph-cachet/carp.sensing-flutter/tree/master/backends/carp_webservices), the request throws an exception with a **400 Bad Request error**.

`_status = await CarpDeploymentService().createStudyDeployment(protocol);`

In the [deployment_request](https://github.com/cph-cachet/carp.sensing-flutter/blob/master/carp_core/lib/carp_deployment/infrastructure/deployment_request.dart) file, I found that the request includes a field with the key **studyDeploymentId** which is necessary for another requests from the deployment service, but not for the **CreateStudyDeployment** class.

I don't make a pull request because I'm not sure how to safely remove this field for this specific request.

Request error response:

`{"statusCode":400,"exception":"dk.cachet.carp.webservices.common.exception.serialization.SerializationException","message":"The deployment service request serializer is not valid. Exception: Encountered unknown key 'studyDeploymentId'.\nUse 'ignoreUnknownKeys = true' in 'Json {}' builder to ignore unknown keys.\nCurrent input: .....ceRoleName\":\"Custom device\"}],\"expectedParticipantData\":[]}}","path":"POST /api/deployment-service"}`

carlos-bailon, 2021-06-09:
I think I have figured out the solution.

I added this to the [CreateStudyDeployment](https://github.com/cph-cachet/carp.sensing-flutter/blob/master/carp_core/lib/carp_deployment/infrastructure/deployment_request.dart) class:

`@JsonKey(ignore: true)`
 `String studyDeploymentId;`

And then re-built the serialization files, so that this field is not included when an object of this class is converted to JSON and embedded in the request, but the field is still usable by the rest of the classes.

Since my knowledge of flutter and dart is very limited, is this the appropriate way of solving this issue?

bardram, 2021-06-14:
Yes - this is one way to solve it. Looks good. Do you want to make a PR on this?

carlos-bailon, 2021-06-14:
Done!

closed 2021-06-14 by carlos-bailon