carp-dk/carp.sensing-flutter #225: All app files gone on iOS

sc00n, 2021-11-08:
On almost all our iOS participants we had an issue that after a while of mobile sensing all the app related files (in NSDocumentDirectory on iOS) were gone (removed). As this does not happen to our app when mobile sensing is turned off I suspect it has something to do with the sensing. 

I don't think that there is anything wrong with the CARP code that causes this. I just want to warn the community that this might happen and see if anybody else encountered this problem. We suspect however that it has something to do with gathering a big chunk of data and save many files with sensing data on the phone. 

We now implemented two fixes which seem (until now, fingers crossed) to solve the problem:

1. We make sure all mobile sensing files are uploaded regularly and deleted afterwards. Even if there is no Wifi. We think the problem might be caused by some storage issue. Uploading and deleting the files often may solve this.
2. Giving the most important files in the NSDocumentDirectory the key NSURLIsExcludedFromBackupKey. The reasoning here would be that some files are backed up and removed from the phone when too many files are written in the NSDocumentDirectory.

For now, we have no idea which of both solutions has helped :).

bardram, 2021-11-10:
Thanks for the heads-up. Once we have solved this issues, the solution might go to the wiki documentation where we assembles best practice advice.

 * https://github.com/cph-cachet/carp.sensing-flutter/wiki/5.-Best-Practice

bardram, 2021-11-10:
The `file` data backend allows for zipping files - in my experience, this reduces the json files with a factor 10. See

* https://github.com/cph-cachet/carp.sensing-flutter/wiki/C.-Data-Backends#files

bardram, 2021-11-13:
Actually - I've long been considering to write a data backend using [`sqflite`](https://pub.dev/packages/sqflite). 

@sc00n - would this be something you could do as a PR?

sc00n, 2021-11-16:
We already use zip the files.

We (koenniem and me) would be happy to help with PR's but I know little about databases, so I don't think this task is suited for us. Please let us know when you have something else in mind where we can help!

bardram, 2022-07-15:
I've implemented support for SQLite and it be part of the 0.40.0 release.

bardram, 2022-09-25:
Released with version 0.40.0.

closed 2022-09-25 by bardram