carp-dk/carp.sensing-flutter #243: Null Safety related error on timestamp of datum

luciamontesinos, 2022-01-20:
When trying to use the packages, I'm getting the following error:

```
Error: Method 'toIso8601String' cannot be called on 'DateTime?' because it is potentially null.
../…/lib/audio.g.dart:97
     - 'DateTime' is from 'dart:core'.
    Try calling using ?. instead.
      val['timestamp'] = instance.timestamp.toIso8601String();
```

I am getting this error in the generated files for audio, context, and survey. E.g: 
https://github.com/cph-cachet/carp.sensing-flutter/blob/88c3fa0ec1b07e13a2505035209c0d3b200c755b/packages/carp_audio_package/lib/audio.g.dart#L35

MadsVSChristensen, 2022-01-27:
I am also getting this error inz `carp_health_package`:

```
Error: A value of type 'DateTime?' can't be assigned to a variable of type 'DateTime' because 'DateTime?' is nullable and 'DateTime' isn't.
../…/lib/health_package.g.dart:91
          ? null
          ^
: Warning: Operand of null-aware operation '?.' has type 'DateTime' which excludes null.
../…/lib/health_package.g.dart:104
  writeNotNull('timestamp', instance.timestamp?.toIso8601String());

```

bardram, 2022-03-16:
JSON is update in version 0.33.0 - should fix this.

closed 2022-04-05 by bardram