carp-dk/carp.sensing-flutter #255: Location vs Geolocator

sc00n, 2022-05-04:
The carp_context_package 0.33.0 currently uses the Flutter Location Plugin. However, this currently leads to some problems on my device (Samsung Android 12). I ask for the location every 15 seconds but only get the location every 10 minutes or so (when app is in foreground). The problem is also encountered by other users of the location package: https://github.com/Lyokone/flutterlocation/issues/608 .

Additionally the package seems to be abandoned (no recent updates). Is there a reason for not using the https://pub.dev/packages/geolocator package? If you want I can work on a PR where the context package uses geolocator (or where the context package has new Measures that use the Geolocator package).

sc00n, 2022-05-04:
When I put the following form location_manager in comments, it seems to work again:

```
     /*
      await locationManager.changeSettings(
        accuracy: location.LocationAccuracy.values[
            configuration?.accuracy.index ??
                GeolocationAccuracy.balanced.index],
        distanceFilter: configuration?.distance,
        interval: configuration?.interval.inMilliseconds,
      );

      await locationManager.changeNotificationOptions(
        title: configuration?.notificationTitle ?? 'CARP Location Service',
        subtitle: configuration?.notificationMessage ??
            'The location service is running in the background',
        description: configuration?.notificationDescription ??
            'Background location is on to keep the app up-to-date with your location. '
                'This is required for main features to work properly when the app is not in use.',
        onTapBringToFront: false,
      );*/
```
At least the normal LOCATION works fine, I haven't tested the GEOLOCATION part yet. Every 5 seconds I get a new location (frequency to 5 seconds). 

I'll test tomorrow what the reason for that could be.

sc00n, 2022-05-05:
Ok, I think I know what happens. I have an ImmediateTrigger with ContextSamplingPackage.GEOLOCATION and a PeriodicTrigger for ContextSamplingPackage.LOCATION. Additionally I have some settings for the ContextSamplingPackage.GEOLOCATION in the sampling scheme. 

However (**and this is the problem**), the settings for GEOLOCATION are also used for LOCATION.  As I had a distance>0 in Geolocation and an interval, the periodic trigger from LOCATION doens't work when I don't move (and at best at the interval from GEOLOCATION).  Therefore, when you want the periodic trigger to work you have to set (in GEOLOCATION sampling scheme) the distance to 0 and the interval to a lower duration then the periodic trigger.

bardram, 2022-05-25:
I am updating CAMS to version 0.40.0 where samling configuration is going to be changed slightly. But the above issue still needs to be adressed - the problem is (as far as I can see), that the same `LocationManager` instance is used for both types of measures. 

Will look into it - but will only be fixed in version 0.40.0....

bardram, 2022-09-25:
Version 0.40.0 of CAMS has now been released which is using a `LocationService` for all location-based measures. An example of how this is done is included below (taken from the `carp_context_package` `example.dart`).

```dart
  // Define the online location service and add it as a 'device'
  LocationService locationService = LocationService(
      accuracy: GeolocationAccuracy.high,
      distance: 10,
      interval: const Duration(minutes: 1));
  protocol.addConnectedDevice(locationService);

  // Add a background task that collects location on a regular basis
  protocol.addTriggeredTask(
      IntervalTrigger(period: Duration(minutes: 5)),
      BackgroundTask()
        ..addMeasure(Measure(type: ContextSamplingPackage.LOCATION)),
      locationService);

  // Add a background task that continously collects geolocation and mobility
  // patterns. Delays sampling by 5 minutes.
  protocol.addTriggeredTask(
      DelayedTrigger(delay: Duration(minutes: 5)),
      BackgroundTask()
        ..addMeasure(Measure(type: ContextSamplingPackage.GEOLOCATION))
        ..addMeasure(Measure(type: ContextSamplingPackage.MOBILITY)),
      locationService);
```

Hence, the same location service is used for all location-based measures and the service should hence be configured to take into account all the measures you want to do.

closed 2022-09-25 by bardram