carp-dk/carp.sensing-flutter #278: Location being retrieved even when not mentioned by the study protocol

IgorMatias3, 2022-11-08:
Basically, I removed the mention of "location" or "geolocation" from the protocol, but this is still being retrieved.
Also, I tried to adjust the frequency of this collection (before removing it from the protocol), and it didn't work—always getting data every 500 ms or so (around twice every second, based on the logs).

Tested with a real device—iPhone 12 Pro Max on the latest public stable iOS version.
The location permissions are set to "always allow." I tried downgrading them to "when using," and location data stopped being collected at all (even when on the app's screen).

Find below the protocol.json and the dependencies version.

I hope these information helps you finding the cause.

PROTOCOL ---

`{
    "ownerId": "test@unige.ch",
    "name": "second-test-study",
    "creationDate": "2022-11-08T12:00:00.000000Z",
    "masterDevices": [
     {
      "$type": "dk.cachet.carp.protocols.domain.devices.CustomProtocolDevice",
      "isMasterDevice": true,
      "roleName": "masterphone",
      "samplingConfiguration": {}
     }
    ],
    "connectedDevices": [
        {
        "$type": "dk.cachet.carp.protocols.domain.devices.LocationService",
        "isMasterDevice": false,
        "roleName": "location_service",
        "samplingConfiguration": {},
        "accuracy": "balanced",
        "distance": 10.0,
        "interval": 60000000000
        },
        {
        "$type": "dk.cachet.carp.protocols.domain.devices.WeatherService",
        "isMasterDevice": false,
        "roleName": "weather_service",
        "samplingConfiguration": {},
        "apiKey": "a44242484ff4686854dd8361c2fb48ab"
        }
    ],
    "connections": [],
    "triggers": {
        "0": {
            "$type": "dk.cachet.carp.protocols.domain.triggers.RandomRecurrentTrigger",
            "startTime": {
                "hour": 0,
                "minute": 0,
                "second": 0
            },
            "endTime": {
                "hour": 23,
                "minute": 59,
                "second": 0
            },
                "minNumberOfTriggers": 24,
                "maxNumberOfTriggers": 48,
                "duration": 1000
        },
        "1": {
            "$type": "dk.cachet.carp.protocols.domain.triggers.RandomRecurrentTrigger",
            "startTime": {
                "hour": 8,
                "minute": 0,
                "second": 0
            },
            "endTime": {
                "hour": 20,
                "minute": 0,
                "second": 0
            },
                "minNumberOfTriggers": 3,
                "maxNumberOfTriggers": 8,
                "duration": 1000
        },
        "2": {
            "$type": "dk.cachet.carp.protocols.domain.triggers.ImmediateTrigger"
        },
        "4": {
            "$type": "dk.cachet.carp.protocols.domain.triggers.ImmediateTrigger"
        },
        "5": {
            "$type": "dk.cachet.carp.protocols.domain.triggers.IntervalTrigger",
            "period": 1800000
        }
    },
    "tasks": [
        {
            "$type": "dk.cachet.carp.protocols.domain.tasks.BackgroundTask",
            "name": "Task #5",
            "measures": [
                {
                    "$type": "dk.cachet.carp.protocols.domain.tasks.measures.Measure",
                    "type": "dk.cachet.carp.pedometer"
                },
                {
                    "$type": "dk.cachet.carp.protocols.domain.tasks.measures.Measure",
                    "type": "dk.cachet.carp.battery"
                }
            ]
        },
        {
            "$type": "dk.cachet.carp.protocols.domain.tasks.BackgroundTask",
            "name": "Task #6",
            "measures": [
                {
                    "$type": "dk.cachet.carp.protocols.domain.tasks.measures.Measure",
                    "type": "dk.cachet.carp.device"
                }
            ]
        },
        {
            "$type": "dk.cachet.carp.protocols.domain.tasks.BackgroundTask",
            "name": "Task #7",
            "measures": [
                {
                    "$type": "dk.cachet.carp.protocols.domain.tasks.measures.Measure",
                    "type": "dk.cachet.carp.activity"
                }
            ]
        },
        {
            "$type": "dk.cachet.carp.protocols.domain.tasks.BackgroundTask",
            "name": "Task #9",
            "measures": [
                {
                    "$type": "dk.cachet.carp.protocols.domain.tasks.measures.Measure",
                    "type": "dk.cachet.carp.mobility"
                }
            ]
        },
        {
            "$type": "dk.cachet.carp.protocols.domain.tasks.BackgroundTask",
            "name": "Task #10",
            "measures": [
                {
                    "$type": "dk.cachet.carp.protocols.domain.tasks.measures.Measure",
                    "type": "dk.cachet.carp.weather"
                }
            ]
        }
    ],
    "triggeredTasks": [
        {
        "triggerId": 0,
        "taskName": "Task #5",
        "targetDeviceRoleName": "masterphone"
        },
        {
        "triggerId": 1,
        "taskName": "Task #6",
        "targetDeviceRoleName": "masterphone"
        },
        {
        "triggerId": 2,
        "taskName": "Task #7",
        "targetDeviceRoleName": "masterphone"
        },
        {
        "triggerId": 4,
        "taskName": "Task #9",
        "targetDeviceRoleName": "location_service"
        },
        {
        "triggerId": 5,
        "taskName": "Task #10",
        "targetDeviceRoleName": "weather_service"
        }
    ],
    "expectedParticipantData": [],
    "protocolDescription": {
        "$type": "StudyDescription",
        "title": "second test study",
        "description": "second test study",
        "purpose": "To test sensing coverage",
        "responsible": {
            "$type": "StudyResponsible",
            "id": "kw",
            "name": "Katarzyna Wac",
            "title": "Professor",
            "email": "kw@unige.ch",
            "address": "Route de Drize 7",
            "affiliation": "University of Geneva"
    }
    },
       "description": "The default study testing coverage of most measures. Used in the coverage tests.",
       "dataEndPoint": {
            "$type": "CarpDataEndPoint",
            "type": "CARP",
            "dataFormat": "dk.cachet.carp",
            "bufferSize": 500000,
            "zip": true,
            "encrypt": false,
            "uploadMethod": "DATA_POINT",
            "name": "CARP Production Server",
            "collection": "carp_sensing",
            "deleteWhenUploaded": true
    }
   }`

---

DEPENDENCIES ---

`environment:
  sdk: ">=2.12.0 <3.0.0"

dependencies:
  flutter:
    sdk: flutter
  
  carp_core: ^0.40.3
  carp_mobile_sensing: ^0.40.0
  research_package: any
  carp_context_package: ^0.40.0
  carp_connectivity_package: ^0.40.0
  screen_state: ^2.0.0
  #carp_communication_package: ^0.33.0
  carp_health_package: any
  carp_apps_package: ^0.40.0
  carp_backend: ^0.40.0
  carp_webservices: ^0.40.1
  #redirecting to settings ios/android
  app_settings: ^4.1.8
  adaptive_theme: ^3.1.1
  permission_handler: ^10.2.0

  rxdart: ^0.27.5

  provider: ^6.0.4
  intl: ^0.17.0
  http: ^0.13.5
  cupertino_list_tile: ^0.2.1

  # The following adds the Cupertino Icons font to your application.
  # Use with the CupertinoIcons class for iOS style icons.
  cupertino_icons: ^1.0.5
  shared_preferences: ^2.0.15
  flutter_localizations: 
    sdk: flutter
    version: any
dependency_overrides:
  rxdart: any 
  telephony: ^0.2.0
  battery_plus: any

dev_dependencies:
  flutter_test:
    sdk: flutter
  flutter_launcher_icons: ^0.10.0`

sc00n, 2022-11-12:
When I am debugging I think I have to change the studyId before it also changes the protocol. Otherwise it sometimes still uses the old version (I just started debugging with the udpate of this package, so I might be wrong).

IgorMatias3, 2022-11-13:
Thanks, @sc00n, I'll try that tomorrow or overmorrow. I'll then post here if it worked.

IgorMatias3, 2022-11-30:
@sc00n Update and fix: I could only manage the new protocol to be downloaded to the device after creating a new study from scratch with the updated protocol. For now, this is fixed, but in the future, I will need to find a way to update the protocol and force all devices in the study to download and use it. (closing this issue)

closed 2022-11-30 by IgorMatias3