carp-dk/carp.sensing-flutter #280: IntervalTrigger not working

sc00n, 2022-11-12:
When I use interval trigger it works for a few seconds and then it stops. For example when I have the following code in local_study_protocol_managaer.dart:

```
    protocol.addTriggeredTask(
        IntervalTrigger(period: Duration(milliseconds: 300)),
        BackgroundTask()
          ..addMeasure(Measure(type: SensorSamplingPackage.PEDOMETER)),
        phone);
```

I get the following output:

<details>
<summary>Output</summary>
<br>

```
Performing hot restart...                                               
Restarted application in 1.058ms.
I/flutter (16264): [CAMS INFO] Time zone set to Europe/Brussels
I/flutter (16264): [CAMS INFO] Settings initialized
I/flutter (16264): [CAMS INFO] SensingBLoC initialized
I/flutter (16264): [CAMS INFO] Initializing Sensing - mode: DeploymentMode.LOCAL
I/flutter (16264): ===========================================================
I/flutter (16264):   CARP Mobile Sensing (CAMS) - SmartPhoneClientManager
I/flutter (16264): ===========================================================
I/flutter (16264):   deployment service : SmartphoneDeploymentService
I/flutter (16264):    device controller : DeviceController [6]
I/flutter (16264):            device ID : SP1A.210812.016
I/flutter (16264):    available devices : (Smartphone, LocationService, AirQualityService, ..., PolarDevice, ESenseDevice)
I/flutter (16264): ===========================================================
I/flutter (16264): [CAMS INFO] Adding study to SmartPhoneClientManager - Study - studyDeploymentId: asdf222, deviceRoleName: masterphone
I/flutter (16264): SmartphoneDeploymentController - Study deployment 'asdf222' successfully deployed.
I/flutter (16264): [CAMS INFO] Saving deployment to file '/data/user/0/dk.cachet.carp_mobile_sensing_app/app_flutter/carp/deployments/asdf222/deployment.json'.
I/flutter (16264): [CAMS INFO] Configuring SmartphoneDeploymentController
I/flutter (16264): [CAMS INFO] Initializing device manager, type: dk.cachet.carp.protocols.domain.devices.Smartphone, descriptor.: Smartphone - roleName: masterphone, isMasterDevice: true
I/flutter (16264): [CAMS DEBUG] SmartphoneDeviceManager - setting device status: DeviceStatus.initialized
I/flutter (16264): [CAMS INFO] AppTaskController - Restoring task queue from file '/data/user/0/dk.cachet.carp_mobile_sensing_app/app_flutter/carp/tasks.json'.
I/flutter (16264): [CAMS INFO] AwesomeNotificationController initialized.
I/flutter (16264): [CAMS INFO] Asking for permission for all measure types.
D/permissions_handler(16264): Bluetooth permission missing in manifest
I/flutter (16264): [CAMS INFO] Permissions for Permission.location : PermissionStatus.granted
I/flutter (16264): [CAMS INFO] Permissions for Permission.bluetoothScan : PermissionStatus.granted
I/flutter (16264): [CAMS INFO] Permissions for Permission.locationAlways : PermissionStatus.granted
I/flutter (16264): [CAMS INFO] Permissions for Permission.activity_recognition : PermissionStatus.granted
I/flutter (16264): [CAMS INFO] Permissions for Permission.microphone : PermissionStatus.granted
I/flutter (16264): [CAMS INFO] Permissions for Permission.camera : PermissionStatus.granted
D/permissions_handler(16264): Bluetooth permission missing in manifest
I/flutter (16264): [CAMS INFO] Permissions for Permission.bluetooth : PermissionStatus.denied
I/flutter (16264): [CAMS INFO] Permissions for Permission.bluetoothConnect : PermissionStatus.granted
I/flutter (16264): [CAMS INFO] Initializing SQLiteDataManager...
I/flutter (16264): [CAMS DEBUG] SQLiteDataManager - SQLite DB created
I/flutter (16264): [CAMS INFO] SQLite DB file path : /data/user/0/dk.cachet.carp_mobile_sensing_app/databases/carp-data-2022-11-12-10-11-27-934492Z.db
I/flutter (16264): [CAMS INFO] SmartphoneDeviceManager - Trying to connect to device of type: dk.cachet.carp.protocols.domain.devices.Smartphone and id: SP1A.210812.016
I/flutter (16264): [CAMS DEBUG] SmartphoneDeviceManager - setting device status: DeviceStatus.connected
I/flutter (16264): [CAMS INFO] SmartphoneDeviceManager - Trying to connect to device of type: dk.cachet.carp.protocols.domain.devices.Smartphone and id: SP1A.210812.016
I/flutter (16264): [CAMS DEBUG] SmartphoneDeviceManager - setting device status: DeviceStatus.connected
I/flutter (16264): [CAMS INFO] SmartphoneDeviceManager - Trying to connect to device of type: dk.cachet.carp.protocols.domain.devices.Smartphone and id: SP1A.210812.016
I/flutter (16264): [CAMS DEBUG] SmartphoneDeviceManager - setting device status: DeviceStatus.connected
I/flutter (16264): [CAMS INFO] SmartphoneDeviceManager - Trying to connect to device of type: dk.cachet.carp.protocols.domain.devices.Smartphone and id: SP1A.210812.016
I/flutter (16264): [CAMS DEBUG] SmartphoneDeviceManager - setting device status: DeviceStatus.connected
I/flutter (16264): [CAMS INFO] SmartphoneDeviceManager - Trying to connect to device of type: dk.cachet.carp.protocols.domain.devices.Smartphone and id: SP1A.210812.016
I/flutter (16264): [CAMS DEBUG] SmartphoneDeviceManager - setting device status: DeviceStatus.connected
I/flutter (16264): [CAMS INFO] SmartphoneDeviceManager - Trying to connect to device of type: dk.cachet.carp.protocols.domain.devices.Smartphone and id: SP1A.210812.016
I/flutter (16264): [CAMS DEBUG] SmartphoneDeviceManager - setting device status: DeviceStatus.connected
I/flutter (16264): [CAMS INFO] Initializing StudyDeploymentExecutor - configuration: SmartphoneDeployment - studyDeploymentId: asdf222, device: masterphone, title: CAMS App - Sensing Coverage Study, responsible: Alex B. Christensen,
 deployment: SmartphoneDeployment - studyDeploymentId: asdf222, device: masterphone, title: CAMS App - Sensing Coverage Study, responsible: Alex B. Christensen
I/flutter (16264): [CAMS INFO] Initializing TriggeredTaskExecutor - configuration: TriggeredTask - triggerId: 0, task: Task #5, targetDeviceRoleName: masterphone, deployment: SmartphoneDeployment - studyDeploymentId: asdf222, device
: masterphone, title: CAMS App - Sensing Coverage Study, responsible: Alex B. Christensen
I/flutter (16264): [CAMS INFO] Initializing IntervalTriggerExecutor - configuration: Instance of 'IntervalTrigger', deployment: SmartphoneDeployment - studyDeploymentId: asdf222, device: masterphone, title: CAMS App - Sensing Covera
[... 82 lines cut ...]
I/flutter (16264):  "carp_body": {
I/flutter (16264):   "id": "61c86430-6272-11ed-b223-1b4b39d8795f",
I/flutter (16264):   "timestamp": "2022-11-12T10:11:32.467102Z",
I/flutter (16264):   "step_count": 33339
I/flutter (16264):  }
I/flutter (16264): }
I/flutter (16264): [CAMS DEBUG] SQLiteDataManager - writing data point to SQLite - id: 3, type: dk.cachet.carp.pedometer
I/flutter (16264): [CAMS INFO] Resuming BackgroundTaskExecutor
I/flutter (16264): [CAMS INFO] Resuming PedometerProbe
D/SensorManager(16264): unregisterListener
I/flutter (16264): [CAMS INFO] Pausing BackgroundTaskExecutor
I/flutter (16264): [CAMS INFO] Pausing PedometerProbe
D/SensorManager(16264): registerListener :: 191, step_counter  Non-wakeup, 0, 0,
D/SensorManager(16264): unregisterListener
I/flutter (16264): [CAMS INFO] Pausing BackgroundTaskExecutor
I/flutter (16264): [CAMS WARNING]  Trying to pause a BackgroundTaskExecutor in a state where this cannot be done - state: ExecutorState.paused
I/flutter (16264): [CAMS INFO] Pausing BackgroundTaskExecutor
I/flutter (16264): [CAMS WARNING]  Trying to pause a BackgroundTaskExecutor in a state where this cannot be done - state: ExecutorState.paused
I/flutter (16264): [CAMS INFO] Pausing BackgroundTaskExecutor
I/flutter (16264): [CAMS WARNING]  Trying to pause a BackgroundTaskExecutor in a state where this cannot be done - state: ExecutorState.paused
```

</details>

So I have output for a few seconds, and then it completely stops. The same holds for when I use the intervaltrigger with ContextSamplingPackage.LOCATION or DeviceSamplingPackage.DEVICE .  The app is still open and I see no errors. I used the example app from GitHub.

sc00n, 2022-11-12:
The following also doesn't work:

```
    Measure loc =Measure(type: ContextSamplingPackage.GEOLOCATION)..overrideSamplingConfiguration = PeriodicSamplingConfiguration(
      interval: const Duration(seconds: 2),
      duration: const Duration(seconds: 1),
    );
```

What did work for me was using 

`LocationService locationService = LocationService(interval: Duration(seconds: 10));`

pedometer also kind of sheets with every step. DeviceSamplingPackage.DEVICE I still have to figure out. In general I like it more when I can just use an interval (e.g. every 10 seconds) to measure something. This is mainly because of the gaps and the app being killed. If I have no information I don't know if nothing happened or the app was just killed.

sc00n, 2022-11-15:
I think there is just in general something wrong with the interval trigger. Also the following doesnt work while it is one of the examples in the example app (I only changed the duration for testing purposes).

```
  protocol.addTriggeredTask(
        IntervalTrigger(period: Duration(seconds: 5)),
        BackgroundTask()
          ..addMeasure(Measure(type: ContextSamplingPackage.WEATHER)),
        weatherService);
```

bardram, 2022-11-15:
> When I use interval trigger it works for a few seconds and then it stops. For example when I have the following code in local_study_protocol_managaer.dart:
> 
> ```
>     protocol.addTriggeredTask(
>         IntervalTrigger(period: Duration(milliseconds: 300)),
>         BackgroundTask()
>           ..addMeasure(Measure(type: SensorSamplingPackage.PEDOMETER)),
>         phone);
> ```
> 
> I get the following output:
> 
> So I have output for a few seconds, and then it completely stops. 

The `IntervalTrigger` is intended to be used when sampling a single piece of measure - the so-called "One Time" (OT)measure. This is in contrast to the measure which are "Event Based" (EB). In the [overview of measures](https://github.com/cph-cachet/carp.sensing-flutter/wiki/A.-Measure-Types), you can see which measures are OT and EB. 

Since the `PEDOMETER` measure is a EB, it does not make sense to use an `IntervalTrigger` - this measure collect a step count event each time "some" steps are detected by the phone (when this happens depends on the phone and its step counter hardware). Hence, a `PEDOMETER` measure should be triggered always (`ImmediateTrigger`) or - if you want it to happen with a interval - you can use a [PeriodicTrigger](https://pub.dev/documentation/carp_mobile_sensing/latest/domain/PeriodicTrigger-class.html) which runs periodically with a fixed period.

However, it does not make sense to collect step count on such a short time window ( 300 milliseconds) -- the user will not be able to take any steps in such a short timeframe.  So - I would make something like:

```
 protocol.addTriggeredTask(
        PeriodicTrigger(period: Duration(minutes: 30), duration: Duration(minutes: 10),
        BackgroundTask()
          ..addMeasure(Measure(type: SensorSamplingPackage.PEDOMETER)),
        phone);
```

which would collect step counts for 10 minutes every half hour.

>The same holds for when I use the intervaltrigger with ContextSamplingPackage.LOCATION 
>or DeviceSamplingPackage.DEVICE . The app is still open and I see no errors. I used the example app from GitHub.

I think the problem here is that the sampling time is extremely short - 300 miliseconds - the GPS will never be able to collect any location data in such a short time frame

bardram, 2022-11-15:
> pedometer also kind of shoots with every step

Yes - this is how this sensor works on most phones.

bardram, 2022-11-15:
> I think there is just in general something wrong with the interval trigger. Also the following doesnt work while it is one of the examples in the example app (I only changed the duration for testing [...]

Note that in order to use the Open Weather web service, you need a API key. See the [README](https://pub.dev/packages/carp_context_package) file for documentation of this.  You can also look at the [example](https://github.com/cph-cachet/carp.sensing-flutter/blob/master/packages/carp_context_package/example/lib/example.dart) to see how a `WeatherService` is created and added to the protocol.

sc00n, 2022-11-15:
Thank you for your replies! 

I'll start with your last reply. The problem is that it works once (showing that the API is correctly connected) and then stops. I used exactly the code from the example, with a filled in API key:

```
    // Define the online weather service and add it as a 'device'
    WeatherService weatherService =
    WeatherService(apiKey: '066c4ff8fe6a59b17da65a3f96544b80');
    protocol.addConnectedDevice(weatherService);

    // Add a background task that collects weather every 30 miutes.
    protocol.addTriggeredTask(
        IntervalTrigger(period: Duration(seconds: 5)),
        BackgroundTask()
          ..addMeasure(Measure(type: ContextSamplingPackage.WEATHER)),
        weatherService); 
```

With location the following code doesn't work. It again works once (one output), and then stops.

```
    LocationService locationService = LocationService();
    protocol.addConnectedDevice(locationService);

    // Add a background task that collects location on a regular basis
    protocol.addTriggeredTask(
        IntervalTrigger(period: Duration(seconds: 15)),
        BackgroundTask()
          ..addMeasure(Measure(type: ContextSamplingPackage.GEOLOCATION)),
        locationService);
```

The following however does work:

```
    LocationService locationService = LocationService(interval: Duration(seconds: 15));
    protocol.addConnectedDevice(locationService);

    // Add a background task that collects location on a regular basis
    protocol.addTriggeredTask(
        ImmediateTrigger(),
        BackgroundTask()
          ..addMeasure(Measure(type: ContextSamplingPackage.GEOLOCATION)),
        locationService);
```

Showing that 15 seconds is not too fast to sample but that the IntervalTrigger somehow fails. So for location, it doesn't really matter if IntervalTrigger doesn't work as there is a work around. This can however not be used to the weather measure or others.

sc00n, 2022-11-16:
Ways to reproduce:

1. Import Example app (https://github.com/cph-cachet/carp.sensing-flutter.git)
2. In pubspec.yaml change _research_package_ to _research_package: ^1.2.0_  (otherwise pub get didn't work)
3. Do pub get
4. Go to getStudyProtocol in Sensing/local_study_protocol_mananger.dart
5. Replace everything after protocol.addMasterDevice(phone); with the following code (to focus on one sensor):

```
    LocationService locationService = LocationService();
    protocol.addConnectedDevice(locationService);

    // Add a background task that collects location on a regular basis
    protocol.addTriggeredTask(
        IntervalTrigger(period: Duration(seconds: 15)),
        BackgroundTask()
          ..addMeasure(Measure(type: ContextSamplingPackage.LOCATION)),
        locationService);

    return protocol;
```

6. Run app using the command _Flutter run_ 
7. Click on the play button to start sensing
8. After a couple of seconds, accept location to be accessed (while using app)

I then get the following output after a couple of minutes:

<details>
<summary>Output</summary>

```
An Observatory debugger and profiler on SM G780G is available at: http://127.0.0.1:53607/FCsp_gDDKNE=/
I/flutter ( 9990): [CAMS INFO] Initializing Sensing - mode: DeploymentMode.LOCAL
I/flutter ( 9990): ===========================================================
I/flutter ( 9990):   CARP Mobile Sensing (CAMS) - SmartPhoneClientManager
I/flutter ( 9990): ===========================================================
I/flutter ( 9990):   deployment service : SmartphoneDeploymentService
I/flutter ( 9990):    device controller : DeviceController [6]
I/flutter ( 9990):            device ID : SP1A.210812.016
I/flutter ( 9990):    available devices : (Smartphone, LocationService, AirQualityService, ..., PolarDevice, ESenseDevice)
I/flutter ( 9990): ===========================================================
I/flutter ( 9990): [CAMS INFO] Adding study to SmartPhoneClientManager - Study - studyDeploymentId: 2f1b0280-65a2-11ed-906e-ad0676ea31bd, deviceRoleName: masterphone
I/flutter ( 9990): SmartphoneDeploymentController - Study deployment '2f1b0280-65a2-11ed-906e-ad0676ea31bd' successfully deployed.
The Flutter DevTools debugger and profiler on SM G780G is available at: http://127.0.0.1:9101?uri=http://127.0.0.1:53607/FCsp_gDDKNE=/
I/flutter ( 9990): [CAMS INFO] Saving deployment to file '/data/user/0/dk.cachet.carp_mobile_sensing_app/app_flutter/carp/deployments/2f1b0280-65a2-11ed-906e-ad0676ea31bd/deployment.js
on'.
I/flutter ( 9990): [CAMS INFO] Configuring SmartphoneDeploymentController
I/flutter ( 9990): [CAMS INFO] Initializing device manager, type: dk.cachet.carp.protocols.domain.devices.Smartphone, descriptor.: Smartphone - roleName: masterphone, isMasterDevice: t
rue
I/flutter ( 9990): [CAMS DEBUG] SmartphoneDeviceManager - setting device status: DeviceStatus.initialized
I/flutter ( 9990): [CAMS INFO] Initializing device manager, type: dk.cachet.carp.protocols.domain.devices.LocationService, descriptor.: LocationService - roleName: location_service, is
MasterDevice: false
I/flutter ( 9990): [CAMS DEBUG] LocationServiceManager - setting device status: DeviceStatus.initialized
I/flutter ( 9990): [CAMS INFO] AppTaskController - Restoring task queue from file '/data/user/0/dk.cachet.carp_mobile_sensing_app/app_flutter/carp/tasks.json'.
I/flutter ( 9990): [CAMS WARNING]  AppTaskController - Failed to load task queue - FileSystemException: Cannot open file, path = '/data/user/0/dk.cachet.carp_mobile_sensing_app/app_flu
tter/carp/tasks.json' (OS Error: No such file or directory, errno = 2)
I/flutter ( 9990): [CAMS INFO] AwesomeNotificationController initialized.
W/Android: [Awesome Notifications]( 9990): Channel model 'basic_channel' was not found (ChannelManager:88)
D/Android: [Awesome Notifications]( 9990): Notification channel Basic Notifications created (ChannelManager:214)
W/Android: [Awesome Notifications]( 9990): Channel model 'scheduled_channel' was not found (ChannelManager:88)
D/Android: [Awesome Notifications]( 9990): Notification channel Scheduled Notifications created (ChannelManager:214)
I/flutter ( 9990): [CAMS INFO] Asking for permission for all measure types.
I/BufferQueueProducer( 9990): [SurfaceView - dk.cachet.carp_mobile_sensing_app/dk.cachet.carp_mobile_sensing_app.MainActivity@12851dc@0#1(BLAST Consumer)1](id:270600000001,api:1,p:9990
,c:9990) queueBuffer: queued for the first time.
D/permissions_handler( 9990): Bluetooth permission missing in manifest
D/ViewRootImpl@8de5299[MainActivity]( 9990): Creating frameDrawingCallback nextDrawUseBlastSync=false reportNextDraw=true hasBlurUpdates=false
D/ViewRootImpl@8de5299[MainActivity]( 9990): Creating frameCompleteCallback
I/SurfaceView@12851dc( 9990): uSP: rtp = Rect(0, 0 - 1080, 2256) rtsw = 1080 rtsh = 2256
I/SurfaceView@12851dc( 9990): onSSPAndSRT: pl = 0 pt = 0 sx = 1.0 sy = 1.0
[... 157 lines cut ...]
I/flutter ( 9990):   "user_id": "2f3f7a70-65a2-11ed-ba57-75da8eb74f57",
I/flutter ( 9990):   "start_time": "2022-11-16T11:34:00.678681Z",
I/flutter ( 9990):   "data_format": {
I/flutter ( 9990):    "namespace": "dk.cachet.carp",
I/flutter ( 9990):    "name": "location"
I/flutter ( 9990):   }
I/flutter ( 9990):  },
I/flutter ( 9990):  "carp_body": {
I/flutter ( 9990):   "id": "90cb8e00-65a2-11ed-a2c1-43896f156ddb",
I/flutter ( 9990):   "timestamp": "2022-11-16T11:34:00.671813Z",
I/flutter ( 9990):   "time": "2022-11-16T12:33:41.715",
I/flutter ( 9990):   "latitude": 50.8756757,
I/flutter ( 9990):   "longitude": 4.7075556,
I/flutter ( 9990):   "altitude": 81.9000015258789,
I/flutter ( 9990):   "accuracy": 15.397000312805176,
I/flutter ( 9990):   "speed": 0.0745985358953476,
I/flutter ( 9990):   "speed_accuracy": 0.0,
I/flutter ( 9990):   "heading": 0.0
I/flutter ( 9990):  }
I/flutter ( 9990): }

```
</details>

I have the same problem with other OT measurements.

koenniem, 2022-11-16:
I reproduced this problem uses the steps above on a Xiaomi Mi 10T (Android 11) Lite and also found it only ran once. Additionally, I got a warning:

```
2022-11-16 13:04:45.242 8964-8994/dk.cachet.carp_mobile_sensing_app I/flutter: [CAMS INFO] Pausing BackgroundTaskExecutor
2022-11-16 13:04:45.243 8964-8994/dk.cachet.carp_mobile_sensing_app I/flutter: [CAMS INFO] Pausing LocationProbe
2022-11-16 13:04:47.285 8964-8994/dk.cachet.carp_mobile_sensing_app I/flutter: [CAMS INFO] Pausing LocationProbe
2022-11-16 13:04:47.285 8964-8994/dk.cachet.carp_mobile_sensing_app I/flutter: [CAMS WARNING]  Trying to pause a LocationProbe in a state where this cannot be done - state: ExecutorState.paused
```

I also made sure all permissions were given and battery optimisation was disabled. I then restarted to app to start in a clean slate but the behaviour was the same. Pausing and then restarting sensing (by using the  green button in the app) again produced a single measurement.

bardram, 2022-11-18:
I've just found that the same problem applies for the `PeriodicTriggerExecutor` ... will fix it now.

bardram, 2022-11-18:
Has been fixed now - released as version 0.40.5

closed 2022-11-18 by bardram