carp-dk/carp.sensing-flutter #283: Accelerometer gravity

koenniem, 2022-11-23:
Currently, `accelerometer` and `periodic_accelerometer` contain raw accelerometer measurements that include gravity. However, in analyses this gravity can be problematic as it convolutes any calculation to, for example, classify a distinct type of activity. More concretely, the real value along the x, y, and z-axis cannot be simply measured as these values are added by a (partial) portion of the gravity, based on the phone's rotation. Getting rid of the gravity on the individual x, y, and z-axes has only recently been possibly thanks to the addition of the magnetometer to CARP but this is not a straightforward process. 

I am wondering if there is any reason why CARP uses measurments from the regular accelerometer instead those of those from the [linear accelerometer](https://developer.android.com/reference/android/hardware/Sensor#TYPE_LINEAR_ACCELERATION) on Android or the [userAcceleration](https://developer.apple.com/documentation/coremotion/cmdevicemotion/1616149-useracceleration) on iOS. Measurements from the linear accelerometer are based on the same measurements from the gyroscope but processed internally to remove gravity by including either the gyroscope or the magnetometer. I can imagine it may be due to unavailability on lower level APIs (though it has been available since API 9) or higher battery consumption, but even with these concerns I think there is a strong case to prefer accelerometer measurements without gravity.

Looking forward to hear your thoughts on this.

davwillev, 2022-11-24:
@koenniem I agree. For what it's worth, ResearchKit uses [userAcceleration](https://developer.apple.com/documentation/coremotion/cmdevicemotion/1616149-useracceleration).

bardram, 2022-11-24:
This is because we rely on the [sensors_plus](https://pub.dev/packages/sensors_plus) package which only provides sensor data including gravity.

davwillev, 2022-11-24:
Hi @bardram ,
I just took a quick look the [sensors_plus](https://pub.dev/packages/sensors_plus) package and they say they describe one of their streams, `UserAccelerometerEvent`, as not including gravity.

bardram, 2022-11-25:
Released as part of CAMS 0.40.6.

Now you can write the following in your study protocol:

```
    protocol.addTriggeredTask(
        PeriodicTrigger(
            period: Duration(seconds: 6), duration: Duration(seconds: 2)),
        BackgroundTask(measures: [
          Measure(type: SensorSamplingPackage.USER_ACCELEROMETER)
        ]),
        phone);
```

closed 2022-11-25 by bardram

koenniem, 2022-11-25:
Awesome! Thanks for adding this so quickly!