carp-dk/carp.sensing-flutter #287: Bluetooth measure takes many samples and writes in invalid JSON format

koenniem, 2022-11-24:
### Device / Emulator and OS
Please complete the following information for each phone and/or emulator you're experiencing this bug on:
 - Device: Xiaomi Mi 10T Lite and Samsung S20 FE
 - OS: Android 12

### Describe the bug
When using the Bluetooth measure in `carp_connectivity_package`, the output are cumulative intermediate measurements instead of only the final result. In other words, instead of waiting for results, multiple measurements appear in the output with an increasing number of devices found in the Bluetooth scan. Concretely, the first result is likely to contain no found Bluetooth devices in the vicinity, the second measurement a few, in the third measurement, a few more, et cetera. The total number of measurement on my phone was 12 instead of just 1. 

Additionally, in the case of using a `FileDataEndPoint`, the measurements appear in invalid JSON format (see attachment). 

### To Reproduce
1. Download the example app in apps/carp_mobile_sensing_app
2. In pubspec.yaml change research_package to research_package: ^1.2.0 (otherwise `pub get` doesn't work)
3. Do pub get
4. Go to `getStudyProtocol` in `Sensing/local_study_protocol_mananger.dart`
Replace everything after protocol.addMasterDevice(phone); with the following code (to focus on one sensor):
```dart
    Measure bluetoothMeasure = Measure(type: ConnectivitySamplingPackage.BLUETOOTH);
    bluetoothMeasure.overrideSamplingConfiguration = PeriodicSamplingConfiguration(
      interval: const Duration(seconds: 20),
      duration: const Duration(seconds: 5),
    );

    protocol.addTriggeredTask(
        ImmediateTrigger(),
        BackgroundTask()
          ..addMeasure(bluetoothMeasure),
        phone);

    return protocol;
```` 
5. Run the app
6. After 20 seconds, the number of samples will appear on-screen and in the terminal.
I then get the following output after a couple of minutes:

### Expected behavior
A properly formatted JSON file with exactly one Bluetooth measurement. 

### Actual behavior
<details>
  <summary>Output</summary>

```json
[
{"carp_header":{"study_id":"9498b4f0-69bb-11ed-b196-6fe3b5cd024f","device_role_name":"masterphone","trigger_id":"0","user_id":"94bc1b70-69bb-11ed-be10-eb304a0ea062","start_time":"2022-11-23T13:07:34.882635Z","data_format":{"namespace":"dk.cachet.carp","name":"bluetooth"}},"carp_body":{"id":"cc03cef0-6b2f-11ed-8503-c9d2f73b762a","timestamp":"2022-11-23T13:07:34.879608Z","scan_result":[]}}{"carp_header":{"study_id":"9498b4f0-69bb-11ed-b196-6fe3b5cd024f","device_role_name":"masterphone","trigger_id":"0","user_id":"94bc1b70-69bb-11ed-be10-eb304a0ea062","start_time":"2022-11-23T13:07:34.895827Z","data_format":{"namespace":"dk.cachet.carp","name":"bluetooth"}},"carp_body":{"id":"cc041d10-6b2f-11ed-9862-554bf89cf15e","timestamp":"2022-11-23T13:07:34.881143Z","scan_result":[]}}
,

,
{"carp_header":{"study_id":"9498b4f0-69bb-11ed-b196-6fe3b5cd024f","device_role_name":"masterphone","trigger_id":"0","user_id":"94bc1b70-69bb-11ed-be10-eb304a0ea062","start_time":"2022-11-23T13:07:35.073206Z","data_format":{"namespace":"dk.cachet.carp","name":"bluetooth"}},"carp_body":{"id":"cc214200-6b2f-11ed-a9be-9145482eb747","timestamp":"2022-11-23T13:07:35.072020Z","scan_result":[{"advertisement_name":"","bluetooth_device_id":"17:D8:C8:11:C2:A6","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-82}]}}{"carp_header":{"study_id":"9498b4f0-69bb-11ed-b196-6fe3b5cd024f","device_role_name":"masterphone","trigger_id":"0","user_id":"94bc1b70-69bb-11ed-be10-eb304a0ea062","start_time":"2022-11-23T13:07:35.078492Z","data_format":{"namespace":"dk.cachet.carp","name":"bluetooth"}},"carp_body":{"id":"cc222c60-6b2f-11ed-8223-0dba721fdd3a","timestamp":"2022-11-23T13:07:35.078175Z","scan_result":[{"advertisement_name":"","bluetooth_device_id":"17:D8:C8:11:C2:A6","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-82},{"advertisement_name":"","bluetooth_device_id":"64:F7:12:93:4C:30","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-43}]}}{"carp_header":{"study_id":"9498b4f0-69bb-11ed-b196-6fe3b5cd024f","device_role_name":"masterphone","trigger_id":"0","user_id":"94bc1b70-69bb-11ed-be10-eb304a0ea062","start_time":"2022-11-23T13:07:35.081574Z","data_format":{"namespace":"dk.cachet.carp","name":"bluetooth"}},"carp_body":{"id":"cc22a190-6b2f-11ed-98cf-71ef64eca64a","timestamp":"2022-11-23T13:07:35.081305Z","scan_result":[{"advertisement_name":"","bluetooth_device_id":"17:D8:C8:11:C2:A6","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-82},{"advertisement_name":"","bluetooth_device_id":"64:F7:12:93:4C:30","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-43},{"advertisement_name":"","bluetooth_device_id":"30:67:8A:65:69:8C","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-46}]}}{"carp_header":{"study_id":"9498b4f0-69bb-11ed-b196-6fe3b5cd024f","device_role_name":"masterphone","trigger_id":"0","user_id":"94bc1b70-69bb-11ed-be10-eb304a0ea062","start_time":"2022-11-23T13:07:35.087396Z","data_format":{"namespace":"dk.cachet.carp","name":"bluetooth"}},"carp_body":{"id":"cc238bf0-6b2f-11ed-9964-a96ba5434091","timestamp":"2022-11-23T13:07:35.086825Z","scan_result":[{"advertisement_name":"","bluetooth_device_id":"17:D8:C8:11:C2:A6","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-82},{"advertisement_name":"","bluetooth_device_id":"64:F7:12:93:4C:30","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-43},{"advertisement_name":"","bluetooth_device_id":"30:67:8A:65:69:8C","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-46},{"advertisement_name":"","bluetooth_device_id":"33:1B:BD:D6:B9:F7","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-91}]}}{"carp_header":{"study_id":"9498b4f0-69bb-11ed-b196-6fe3b5cd024f","device_role_name":"masterphone","trigger_id":"0","user_id":"94bc1b70-69bb-11ed-be10-eb304a0ea062","start_time":"2022-11-23T13:07:35.091241Z","data_format":{"namespace":"dk.cachet.carp","name":"bluetooth"}},"carp_body":{"id":"cc242830-6b2f-11ed-998e-695bb6a6d872","timestamp":"2022-11-23T13:07:35.090956Z","scan_result":[{"advertisement_name":"","bluetooth_device_id":"17:D8:C8:11:C2:A6","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-82},{"advertisement_name":"","bluetooth_device_id":"64:F7:12:93:4C:30","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-43},{"advertisement_name":"","bluetooth_device_id":"30:67:8A:65:69:8C","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-46},{"advertisement_name":"","bluetooth_device_id":"33:1B:BD:D6:B9:F7","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-91},{"advertisement_name":"","bluetooth_device_id":"20:90:37:52:A5:AD","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-70}]}}{"carp_header":{"study_id":"9498b4f0-69bb-11ed-b196-6fe3b5cd024f","device_role_name":"masterphone","trigger_id":"0","user_id":"94bc1b70-69bb-11ed-be10-eb304a0ea062","start_time":"2022-11-23T13:07:35.095525Z","data_format":{"namespace":"dk.cachet.carp","name":"bluetooth"}},"carp_body":{"id":"cc24c470-6b2f-11ed-bb8d-57f5cc5b1a36","timestamp":"2022-11-23T13:07:35.095259Z","scan_result":[{"advertisement_name":"","bluetooth_device_id":"17:D8:C8:11:C2:A6","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-82},{"advertisement_name":"","bluetooth_device_id":"64:F7:12:93:4C:30","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-43},{"advertisement_name":"","bluetooth_device_id":"30:67:8A:65:69:8C","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-46},{"advertisement_name":"","bluetooth_device_id":"33:1B:BD:D6:B9:F7","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-91},{"advertisement_name":"","bluetooth_device_id":"20:90:37:52:A5:AD","bluetooth_device_name":"","bluetooth_device_type":"unknown","connectable":false,"rssi":-70},{"advertisement_name":"","bluetooth_device_id":"70:06:BA:EC:A1:32","bluetooth_device_name":"","bluetooth_device_type":"unknown","co [... cut ...]
    • Flutter extension can be installed from:
       https://marketplace.visualstudio.com/items?itemName=Dart-Code.flutter

[√] Connected device (3 available)
    • Windows (desktop) • windows • windows-x64    • Microsoft Windows [Version 10.0.19044.2251]
    • Chrome (web)      • chrome  • web-javascript • Google Chrome 107.0.5304.107
    • Edge (web)        • edge    • web-javascript • Microsoft Edge 107.0.1418.56

[√] HTTP Host Availability
    • All required HTTP hosts are available

• No issues found!
```
</details>

bardram, 2022-12-06:
You're right. Looking at this now.

bardram, 2022-12-06:
Released as version 0.40.1 of the carp_connectivity_package package.

closed 2022-12-06 by bardram