carp-dk/carp.sensing-flutter #292: Question: is there an easy way to change the protocol.

sc00n, 2022-12-02:
Lets say I want to completely change the protocol in the middle of a study. I have protocolPast and protocolFuture on the phone. What is the easiest way to do this in the app? I want to

1. Stop all sampling from protocolPast
2. initiate protocolFuture
3. Start all sampling from protocolFuture

Just running await Sensing().initialize() again with a new protocol clearly doesn't work. I'm probably ovelrooking something but I couldnt find it.

sc00n, 2022-12-02:
Changing the name (deployment id) of study seems to work :

```
_status = await SmartphoneDeploymentService().createStudyDeployment(
  protocol,
  studyName,
);

bloc.studyDeploymentId = studyName;
```

will test it some more

bardram, 2022-12-06:
There is this option:

```
  // Alternatively mark the deplyment as changed - calling hasChanged()
  // this will force a restart of the entire sampling
  controller.deployment?.hasChanged();
```

This should notify all executors (probes etc.) that the deployment has changed and sampling is to be restarted. But.... as far as I can see, this actually does not work...

bardram, 2022-12-06:
If this is an entirely **new** protocol (and not an adaptation of the existing one), I would simply create a new deployment like this:

```

    // Stop the old controller
    oldController?.stop();

    // create the new study
    study = await client.addStudyProtocol(protocolFuture);

    // Get a new study controller and try to deploy the study.
    controller = client.getStudyRuntime(study!);
    await controller?.tryDeployment(useCached: false);

    // Configure the new controller.
    await controller?.configure();
```

sc00n, 2023-01-07:
This worked for me! I also added  
`controller?.start(false);
`
at the end. But I dont know if this is necsessary.

bardram, 2023-01-18:
> This worked for me! I also added
> `controller?.start(false); `
> at the end. But I dont know if this is necsessary.

Yes - you would need to "stop" the old controller and "start" the new one.

closed 2023-01-18 by bardram