carp-dk/carp.sensing-flutter #293: Wifi measure does not work

koenniem, 2022-12-05:
### Device / Emulator and OS
 - Device: Xiaomi Mi 10T Lite and Samsung S20 FE
 - OS: Android 12

### Describe the bug
When implementing a Wifi measure in the example app (using an `ImmediateTrigger`), there is not output.

### To Reproduce
1. Download the example app in `apps/carp_mobile_sensing_app`.
2. In pubspec.yaml change research_package to research_package: ^1.2.0 (otherwise `pub get` doesn't work)
3. Do pub get
4. Go to `getStudyProtocol` in `Sensing/local_study_protocol_mananger.dart`
Replace everything after protocol.addMasterDevice(phone); with the following code (to focus on one sensor):
```dart
    protocol.addTriggeredTask(
        ImmediateTrigger(),
        BackgroundTask()
          ..addMeasures([
            Measure(type: ConnectivitySamplingPackage.WIFI),
          ]),
        phone);

    return protocol;
```` 
5. Run the app

### Expected behavior
After 10 minutes (the default for `WifiMeasure`), a list of Wi-Fi networks should appear in the teriminal.

### Actual behavior
There are no measurements.

sc00n, 2022-12-06:
I got it to work using the following code:

```
      Measure wifiMeasure = Measure(type: ConnectivitySamplingPackage.WIFI);
      wifiMeasure..overrideSamplingConfiguration = PeriodicSamplingConfiguration(
        interval: Duration(seconds: 10),
        duration: Duration(seconds: 2),
      );

      protocol.addTriggeredTask(
          ImmediateTrigger(),
          BackgroundTask()
            ..addMeasure(wifiMeasure),
          phone);
```

bardram, 2022-12-06:
Good point. I've added this to the README file.

bardram, 2022-12-06:
Released as version 0.40.1 of the carp_connectivity_package package.

closed 2022-12-06 by bardram