carp-dk/carp.sensing-flutter #301: Timezone question

sc00n, 2023-01-07:
Is there a way to add the time zone of the participant in the header? The time in the header (at this actual moment) reads 

"start_time": "2023-01-07T20:24:52.266973Z"

But it is 21:24:52 here and I am not in Zulu time zone (but in Alpha time zone aka UTC+1). Its fine if the start_time is just in UTC time, but it would be nice if I also knew the timezone.

bardram, 2023-01-18:
I'm not sure what the issue is? The reason to use zulu time is that you can always convert a zulu time to the local time zone. See e.g. the [timezone](https://pub.dev/packages/timezone) package.

sc00n, 2023-01-19:
When the data is still on the phone you can convert it back to local time. However, when the data is exported away from the phone, this is harder for the researcher. Moreover, it is exactly then that we need to convert it back to local time, for example to link it with other external data (e.g. ESM questionnaire data or data from other sensors and other apps). The problem is that you don't know the local time zones of all participants.  For example: 

- When you have participants spread all over the world
- When some of your participants are travelling during the study 

When you are also gathering locating during the study, you probably could infer the time zone. In some studies we however will not want to gather location data (e.g. for privacy reasons).

bardram, 2023-02-08:
ok - I understand. My main problem with this PR is that the `DataPoint` format incl. the header complies to the API of the CARP web services data upload endpoint. Hence, if we start adding new fields to the data point header, we need to upgrade the backend too. 

I know that you probably do not use the CARP backend. So I will look for another solution to this problem.

bardram, 2023-02-13:
I will add a new data type for time zone - this is IMO the best solution. Also need to collect it on the phone.

bardram, 2023-02-13:
In CAMS v. 0.40.13 you can write a protocol that include collection of time zone information. Examples include:

```
    // Collect timezone info every time the app restarts.
    protocol.addTriggeredTask(
        ImmediateTrigger(),
        BackgroundTask(measures: [
          Measure(type: DeviceSamplingPackage.TIMEZONE),
        ]),
        phone);
```

And you can also add it to an `AppTask` so that you know in which time zone the survey was filled in:

```
    // Add an app task 2 minutes after deployment and make a notification.
    //
    // This App Task is added for demo purpose and you should see notifications
    // on the phone. However, nothing will happen when you click on it.
    // See the PulmonaryMonitor demo app for a full-scale example of how to use
    // the App Task model.
    //
    // Note also that the timezone measure is added. This entails that timezone
    // information is collected when the user 'executes' this app task.
    protocol.addTriggeredTask(
        ElapsedTimeTrigger(
          elapsedTime: const Duration(minutes: 2),
        ),
        AppTask(
          measures: [Measure(type: DeviceSamplingPackage.TIMEZONE)],
          type: BackgroundSensingUserTask.ONE_TIME_SENSING_TYPE,
          title: "Elapsed Time - 2 minutes",
          notification: true,
        ),
        phone);
```

sc00n, 2023-03-14:
Thanks! This seems to work.

closed 2023-03-14 by sc00n