carp-dk/carp.sensing-flutter #308: weather probe not working when air quality is included

sc00n, 2023-01-16:
Weather seems to work ok when I use all sensors excluding air quality. Once I add air quality I don't get anything anymore from weather.

code (with sampling every 10s. In real experiment the sampling will obviously be at a lower frequency).

```
 AirQualityService airService =
      AirQualityService(apiKey: APIAIR);
      protocol.addConnectedDevice(airService);

      protocol.addTriggeredTask(
          IntervalTrigger(period: Duration(seconds: 10)),
          BackgroundTask()
            ..addMeasure(Measure(type: ContextSamplingPackage.AIR_QUALITY)),
          airService);

WeatherService weatherService =
      WeatherService(apiKey: APIWEATHER);
      protocol.addConnectedDevice(weatherService);
      
      protocol.addTriggeredTask(
          IntervalTrigger(period: Duration(seconds: 10)),
          BackgroundTask()
            ..addMeasure(Measure(type: ContextSamplingPackage.WEATHER)),
          weatherService);
```

bardram, 2023-01-18:
Strange - will investigate.

bardram, 2023-02-13:
I cannot reproduce this error. When I configure a protocol to sample both weather and air quality on a regular basis (`IntervalTrigger`) it seems to work fine. 

Two things to note:

* make sure that you're on the latest v. 0.40.12 version of CAMS (previously there was a bug in the `IntervalTrigger`)
* check you weather API key - if you're using the CACHET api key (ending in `844a`) it is on a regular basis blocked due to over-use

bardram, 2023-12-18:
This is still a problem - reopening...

[coverage.pdf](https://github.com/cph-cachet/carp.sensing-flutter/files/13703511/coverage.pdf)