carp-dk/carp.sensing-flutter #320: Impossible to scan for Bluetooth devices on iOS when screen is off or app is in bacground.

sc00n, 2023-03-14:
According to iOS documentation and forums (e.g. https://developer.apple.com/forums/thread/652592), it is impossible to do a general Bluetooth scan when the screen is off or the app is in background. This will simply result in an empty scan. We also see this in our data. This probably should be added to the documentation of the carp_connectivity_package (https://pub.dev/packages/carp_connectivity_package). 

It should however be possible to scan for specific devices when the screen is off. If I find the time, I can try to see if this can be done in flutter_blue_plus . Many researchers want to use the Bluetooth scan to test if a participant is near their partner. For this, a scan to specific device would suffice. (However, I don't expect to find enough time for a deep dive  in the coming months :) )

bardram, 2023-04-04:
Thanks, @sc00n. As a starter, I will incorporate this into the documentation. 

If you come up with additional input, let me know.

sc00n, 2023-09-21:
After looking at this again, I think this is also true for Android (certainly for newer Android versions). So for both Android and iOS , the scan results return empty when the screen is off (most of the times in sensing I guess). This makes the ConnectivitySamplingPackage.BLUETOOTH sensor rather useless . In both Android and iOS it should however be possible to scan for the presence of a specific Bluetooth device, also when the screen is off and the app is in background.

--> there should be a new bluetooth sensor in the ConnectivitySamplingPackage that scans for a specific (set of) device(s).