carp-dk/carp.sensing-flutter #321: Impossible to build for iOS

sc00n, 2023-05-09:
It is impossible to build a flutter project when you include this package.

Way to reproduce:

- Create new empty flutter project
- add _carp_mobile_sensing: 0.40.13_ and do pub get
- try to run the app

In that case, this is my output:

```
Could not build the precompiled application for the device.
Swift Compiler Error (Xcode): Using bridging headers with module interfaces is unsupported

Uncategorized (Xcode): Command SwiftDriver emitted errors but did not return a nonzero exit code to indicate failure

Error (Xcode): no such module '__ObjC'
/Users/merijnmestdagh/Library/Developer/Xcode/DerivedData/Runner-gvgebmsecfctmcftbygkytvpcngb/Build/Intermediates.noindex/Runner.bu
ild/Debug-iphoneos/Runner.build/Objects-normal/arm64/Runner.private.swiftinterface:9:18

Error (Xcode): failed to verify module interface of 'Runner' due to the errors above; the textual interface may be broken by
project issues or a compiler bug
/Users/merijnmestdagh/Library/Developer/Xcode/DerivedData/Runner-gvgebmsecfctmcftbygkytvpcngb/Build/Intermediates.noindex/Runner.bu
ild/Debug-iphoneos/Runner.build/Objects-normal/arm64/Runner.private.swiftinterface:0:0

Error launching application on Merijn’s iPhone.
```

Obviously without the CARP package, the empty default Flutter app runs fine.

The problem also occurs with other CARP packages (e.g. _carp_connectivity_package: ^0.40.1_)

I currently have XCode Version 14.3. A colleague of mine said that the problem disappears when you roll back to XCode 14.2. A month ago, I didn't have this problem yet (same CARP versions). This also points in the direction that something on Apple/XCode has changed.

bardram, 2023-05-15:
I very rarely work on iOS / XCode, so I have not encountered this problem. I will have @LarsRefsgaard look into this - he is our iOS guy around here.

LarsRefsgaard, 2023-05-16:
So the [release notes for XCode 14.3](https://developer.apple.com/documentation/xcode-release-notes/xcode-14_3-release-notes) mention that something was changed in terms of modules... might be that? I'll investigate further in the near future.

LarsRefsgaard, 2023-09-06:
It seems not. I tried making a new project and `flutter pub add carp_mobile_sensing` and I am still unable to build.

majornuts, 2023-09-06:
[image]
my fork build android and ios now, but the ios app that gets build is stuck in eternal loading mode. 💩 
the main take away was bumping the ios version, and disabling the distributed lib build setting.

sc00n, 2023-09-07:
I also find that when I go to build settings, and set 'build libraries for distribution' to no, I can run the app. I guess however that this means that I cannot distribute the app anymore on the app store? This would make this solution not so useful for me.

(probably related problem of which I don't understand everything as I don't know a lot about iOS: https://developer.apple.com/forums/thread/120352)

majornuts, 2023-09-07:
Setting name: BUILD_LIBRARY_FOR_DISTRIBUTION

Ensures that your libraries are built for distribution. For Swift, this enables support for library evolution and generation of a module interface file

My gut feeling tells me it's a library that is non null safe that cascades, or a versioning problem with carp libraries.

sc00n, 2023-09-07:
Ok, so I did some digging! I looked at all the packages that are imported in carp_mobile_sensing. The problem seems to be with awesome_notifications. When I add `awesome_notifications: ^0.7.0` in an empty flutter project, I get the same error.

sc00n, 2023-09-07:
awesome_notifications has indeed the following configuration steps:

```

... and set the following options:

In Runner Target:

Build libraries for distribution => NO
Only safe API extensions => NO
iOS Deployment Target => 11 or greater
In all other Targets:

Build libraries for distribution => NO
Only safe API extensions => YES
```

see https://pub.dev/packages/awesome_notifications

These settings look a bit dodgy... Is awesome_notifications necessary? I also see that they don't really work together well with firebase_messaging, which I certainly need for my app.

LarsRefsgaard, 2023-09-08:
> awesome_notifications has indeed the following configuration steps:
>
> ```
>
> ... and set the following options:
>
> In Runner Target:
>
> Build libraries for distribution => NO
> Only safe API extensions => NO [...]

Have you tried dependency overriding `awesome_notifications` with their prerelease @sc00n ?
We use it to send notifications for tasks. Maybe @bardram can comment on this? Would it be possible to refactor the `NotificationController` to a separate package or something?

LarsRefsgaard, 2023-09-09:
We initially used `flutter_local_notifications` but switched away from it in favour of `awesome_notifications`, but it seems that that might not have been the best idea. @bardram might be able to give some input on exactly why, and if it would be easy to revert back? The advantage of `flutter_local_notifications` seems to be that it is maintained by more than just one single person, who is not that active, looking at their respective repos:
https://github.com/rafaelsetragni/awesome_notifications
https://pub.dev/packages/flutter_local_notifications

Also `flutter_local_notifications` has a Flutter favourite badge, for what it's worth.

majornuts, 2023-09-09:
[Swiftin page link](https://developer.apple.com/documentation/xcode-release-notes/xcode-11_2-release-notes#Swift)
[Known Issuesin page link](https://developer.apple.com/documentation/xcode-release-notes/xcode-11_2-release-notes#Known-Issues)
If a module is built with BUILD_LIBRARIES_FOR_DISTRIBUTION and contains a public type with the same name as the module itself, clients will fail to import the module. (19481048) (FB5863238)

Workaround: Rename either the type or the module to remove the conflict.

LarsRefsgaard, 2023-09-13:
@sc00n 
I have made a branch and a PR (#334 ) to remove `awesome_notifications`, so then I can discuss it with @bardram when he is home next week. For now you can do:
```
dependency_overrides:
  carp_mobile_sensing:
    git:
      url: https://github.com/cph-cachet/carp.sensing-flutter.git
      ref: LarsRefsgaard/issue321
      path: carp_mobile_sensing/
```
in your `pubspec.yaml`.

LarsRefsgaard, 2023-10-03:
#334 is merged into develop-1.1.0 and released as carp_mobile_sensing 1.3.2, closing issue.

PR #334 (2023-09-20): Remove awesome_notifications
Closes #321. 
It was causing issues building for iOS because awesome_notifications has `build for distribution` disabled for some reason. We revert back to flutter local notifications which also is a flutter recommended package.
closed 2023-10-03 by LarsRefsgaard