carp-dk/carp.sensing-flutter #325: Polar package needs upgrading in carp_polar_package

LarsRefsgaard, 2023-06-27:
I get this version solving error in the study app:

```
➜ flutter pub upgrade
Resolving dependencies... (3.8s)
Because every version of carp_polar_package from git depends on polar ^3.4.0 which depends on device_info_plus ^8.0.0, every version of carp_polar_package from git requires device_info_plus ^8.0.0.
And because every version of carp_mobile_sensing from git depends on device_info_plus ^9.0.0, carp_polar_package from git is incompatible with carp_mobile_sensing from git.
So, because carp_study_app depends on both carp_mobile_sensing from git and carp_polar_package from git, version solving failed.
```
https://github.com/cph-cachet/carp.sensing-flutter/blob/develop-1.1.0/packages/carp_polar_package/pubspec.yaml#L20
Perhaps it is also a good idea to keep up with new releases here so we get the latest features and bug fixes.

LarsRefsgaard, 2023-07-20:
Now I am getting this error after running `pub upgrade`:
```
➜ pod install
Analyzing dependencies
[!] CocoaPods could not find compatible versions for pod "PolarBleSdk":
  In snapshot (Podfile.lock):
    PolarBleSdk (= 3.3.6, ~> 3.3.6)

  In Podfile:
    polar (from `.symlinks/plugins/polar/ios`) was resolved to 0.0.1, which depends on
      PolarBleSdk (~> 5.1.0)```

bardram, 2023-08-29:
Is fixed as part of the `1.1.x` upgrade.

closed 2023-08-29 by bardram