carp-dk/carp.sensing-flutter #332: Health Sampling Package upgrade

bardram, 2023-08-27:
The Health Sampling Package needs to be upgraded to handle the `health` plugin as a service. Incl. permissions.

bardram, 2023-08-30:
The health package is now updated. It now runs a `HealthService` which can be added to a protocol like this:

``` 
  // First add this smartphone.
  final phone = Smartphone();
  protocol.addPrimaryDevice(phone);

  // Define which health types to collect.
  var healthDataTypes = [
    HealthDataType.WEIGHT,
    HealthDataType.EXERCISE_TIME,
    HealthDataType.STEPS,
    HealthDataType.SLEEP_ASLEEP,
  ];

  // Create and add a health service (device)
  final healthService = HealthService(types: healthDataTypes);
  protocol.addConnectedDevice(healthService, phone);

  // Automatically collect the set of health data every hour.
  //
  // Note that the [HealthSamplingConfiguration] is a [HistoricSamplingConfiguration]
  // which samples data back in time until last time, data was sampled.
  protocol.addTaskControl(
      PeriodicTrigger(period: Duration(minutes: 60)),
      BackgroundTask()
        ..addMeasure(Measure(type: HealthSamplingPackage.HEALTH)
          ..overrideSamplingConfiguration =
              HealthSamplingConfiguration(healthDataTypes: healthDataTypes)),
      healthService);

``` 

I've added this example protocol to the CAMS Demo App and the health service is shown and is available.

HOWEVER, I cannot make it run on either Android or iOS:

* Android : Google Fit is a nightmare to set up. And when using Health Connect, it requires JDK API level 34, which causes the `shared_preferences` to fail....
* iOS: I can't compile the CAMS Demo App to iOS due to all sorts of pod problems....

So - the health package hasn't really been tested and seen working - yet.

As usual - all of this is on the `develop-1-1-0` branch.

LarsRefsgaard, 2023-11-24:
Working on adding this to the https://github.com/cph-cachet/carp-studies-app .

Clicking on a service and doing this:
```
    switch (service.type) {
      case HealthService.DEVICE_TYPE:
        await service.deviceManager.requestPermissions();
        await service.deviceManager.connect();
```

Getting this log, even after clicking turn on all and done on the native popup:
```
[log] [CAMS INFO] HealthServiceManager - Requesting permissions for device of type: HealthService and id: Apple Health
[log] [CAMS INFO] HealthServiceManager - Checking permissions for device of type: HealthService and id: Apple Health
[log] [CAMS DEBUG] HealthServiceManager - Permission of all permissions: false
[log] [CAMS WARNING] HealthServiceManager has not the permissions required to connect. Call requestPermissions() before calling connect.
```

bardram, 2024-02-13:
Fixed in health package 2.9.3

closed 2024-02-13 by bardram