carp-dk/carp.sensing-flutter #356: Native Geofencing Capabilities

flodaniel, 2024-01-05:
Hello,

I just discovered this project and am amazed! It is amazing.
Looking at the implementation of the geofencing, I can see that the `carp_context_package` relies on the `location` package and its `onLocationChanged` method. This means geofencing will drain much more battery than necessary as it could utilize the native capabilities of Android and iOS. 
I was looking into other packages that allow my app to react to geofencing events if when the app is fully terminated (killed in task switcher) and this would only be possible if the native geofencing is used. From a study and research perspective this could allow a more granular start and stop behavior of a study to avoid draining the battery throughout the day. 

Here is the android documentation: https://github.com/cph-cachet/carp.sensing-flutter/tree/version-0.40.x/apps/carp_mobile_sensing_app 
And here is iOS: https://developer.apple.com/documentation/corelocation/monitoring_the_user_s_proximity_to_geographic_regions

bardram, 2024-01-18:
Thanks for the input. We have been considering using the native geofencing API, but never came about to do this. But definitely on the todo list. 

We need to find a suitable Flutter plugin that supports this, though. Any input on this?

flodaniel, 2024-01-18:
I only found this one, which is not open source: https://pub.dev/packages/flutter_background_geolocation

There does not seem to be any other that correctly implements the native APIs.