carp-dk/carp.sensing-flutter #384: PeriodicTrigger not working

sc00n, 2024-04-23:
When I use a periodic trigger, for device info or weather, it only works once.

Code:

```
   protocol.addTaskControl(
        PeriodicTrigger(period: Duration(seconds: 5)),
        BackgroundTask()
          ..addMeasure(Measure(type: DeviceSamplingPackage.DEVICE_INFORMATION)),
        phone);
```

output:

```
[{"sensorStartTime":1713881375846631,"data":{"__type":"dk.cachet.carp.triggeredtask","triggerId":3,"taskName":"Task #10","destinationDeviceRoleName":"Location Service","control":"Start"}},
{"sensorStartTime":1713881380867767,"data":{"__type":"dk.cachet.carp.triggeredtask","triggerId":4,"taskName":"Task #11","destinationDeviceRoleName":"Primary Phone","control":"Start"}},
{"sensorStartTime":1713881380871224,"data":{"__type":"dk.cachet.carp.deviceinformation","platform":"iOS","deviceId":"F5AA8264-19AA-4A03-814B-32943582AA32","hardware":"iPhone11,2","deviceName":"iPhone","deviceManufacturer":"Apple","deviceModel":"iPhone","operatingSystem":"iOS","deviceData":{"systemName":"iOS","isPhysicalDevice":"true","utsname":{"release":"23.4.0","sysname":"Darwin","nodename":"localhost","machine":"iPhone11,2","version":"Darwin Kernel Version 23.4.0: Fri Mar  8 23:30:38 PST 2024; root:xnu-10063.102.14~67/RELEASE_ARM64_T8020"},"model":"iPhone","localizedModel":"iPhone","systemVersion":"17.4.1","name":"iPhone","identifierForVendor":"F5AA8264-19AA-4A03-814B-32943582AA32"}}},
{"sensorStartTime":1713881385864716,"data":{"__type":"dk.cachet.carp.triggeredtask","triggerId":4,"taskName":"Task #11","destinationDeviceRoleName":"Primary Phone","control":"Start"}},
{"sensorStartTime":1713881390863644,"data":{"__type":"dk.cachet.carp.triggeredtask","triggerId":4,"taskName":"Task #11","destinationDeviceRoleName":"Primary Phone","control":"Start"}},
{"sensorStartTime":1713881395863246,"data":{"__type":"dk.cachet.carp.triggeredtask","triggerId":4,"taskName":"Task #11","destinationDeviceRoleName":"Primary Phone","control":"Start"}},
{"sensorStartTime":1713881400863745,"data":{"__type":"dk.cachet.carp.triggeredtask","triggerId":4,"taskName":"Task #11","destinationDeviceRoleName":"Primary Phone","control":"Start"}},
{"sensorStartTime":1713881405864503,"data":{"__type":"dk.cachet.carp.triggeredtask","triggerId":4,"taskName":"Task #11","destinationDeviceRoleName":"Primary Phone","control":"Start"}},
{"sensorStartTime":1713881410863566,"data":{"__type":"dk.cachet.carp.triggeredtask","triggerId":4,"taskName":"Task #11","destinationDeviceRoleName":"Primary Phone","control":"Start"}},
{"sensorStartTime":1713881415863062,"data":{"__type":"dk.cachet.carp.triggeredtask","triggerId":4,"taskName":"Task #11","destinationDeviceRoleName":"Primary Phone","control":"Start"}},
{"sensorStartTime":1713881420863762,"data":{"__type":"dk.cachet.carp.triggeredtask","triggerId":4,"taskName":"Task #11","destinationDeviceRoleName":"Primary Phone","control":"Start"}},
{"sensorStartTime":1713881425863283,"data":{"__type":"dk.cachet.carp.triggeredtask","triggerId":4,"taskName":"Task #11","destinationDeviceRoleName":"Primary Phone","control":"Start"}},
{"sensorStartTime":1713881430862956,"data":{"__type":"dk.cachet.carp.triggeredtask","triggerId":4,"taskName":"Task #11","destinationDeviceRoleName":"Primary Phone","control":"Start"}},
{"sensorStartTime":1713881435862230,"data":{"__type":"dk.cachet.carp.triggeredtask","triggerId":4,"taskName":"Task #11","destinationDeviceRoleName":"Primary Phone","control":"Start"}},
{"sensorStartTime":1713881440862869,"data":{"__type":"dk.cachet.carp.triggeredtask","triggerId":4,"taskName":"Task #11","destinationDeviceRoleName":"Primary Phone","control":"Start"}},
{"sensorStartTime":1713881445862127,"data":{"__type":"dk.cachet.carp.triggeredtask","triggerId":4,"taskName":"Task #11","destinationDeviceRoleName":"Primary Phone","control":"Start"}},
{"sensorStartTime":1713881450862157,"data":{"__type":"dk.cachet.carp.triggeredtask","triggerId":4,"taskName":"Task #11","destinationDeviceRoleName":"Primary Phone","control":"Start"}},
{"sensorStartTime":1713881455861267,"data":{"__type":"dk.cachet.carp.triggeredtask","triggerId":4,"taskName":"Task #11","destinationDeviceRoleName":"Primary Phone","control":"Start"}},
]
```

As you can see it only once actually got the device info. Its the same for weather (where a periodic trigger is actually useful).

This was tested on iOS with 

```
 carp_core: 1.4.1
  carp_mobile_sensing: 1.5.0
  carp_context_package: ^1.5.0
```

but I also had the problem with previous versions of carp_mobile_sensing (e.g. 1.4.7). 

Will now test on Android.

bardram, 2024-04-23:
It's because the background task executor is never stopped again, and hence cannot be started when already started:

``` 
[log] [CAMS INFO] Starting BackgroundTaskExecutor [1007518798] (started)
[log] [CAMS WARNING] Trying to start a BackgroundTaskExecutor but it is already started. Ignoring this.
```

sc00n, 2024-04-23:
Maybe restart the task? For me the important thing is that we can do something periodically, such as getting a device/weather measurement.

bardram, 2024-04-24:
A way to solve it is to add a duration to the task so it stops automatically after a while:

```
    protocol.addTaskControl(
        PeriodicTrigger(period: const Duration(seconds: 5)),
        BackgroundTask(duration: const IsoDuration(seconds: 2), measures: [
          Measure(type: DeviceSamplingPackage.DEVICE_INFORMATION)
        ]),
        phone);
```

But - there are a couple of issues with this too:

* using an `IsoDuration` instead of a regular Dart `Duration` class seems strange (from a Dart API point-of-view)
* using a `BackgroundTask` for this kind of "on-time" measures seems like a bad design

So - will try to address this.

bardram, 2024-04-24:
carp_core 1.5.0 released solving item #1 above....

bardram, 2024-04-24:
Anyway - another issue is that according to the definition of a `BackgroundTask`:

> The task runs for the specified [duration], or until stopped, or until all measures and/or outputs have completed.

So - I guess that the implementation of the `BackgroundTaskExecutor` does not comply to this contract.

closed 2024-04-25 by bardram