carp-dk/carp.sensing-flutter #389: getting weather (and locatoin) on iOS

sc00n, 2024-04-26:
I found that getting the weather on iOS is not really reliable (I don't mean the weather itself, although also the weather is currently not really reliable here in Belgium). To get the weather CARP first needs to get a location in the weather probe:

`final loc = await LocationManager().getLocation()`

It however often gets stuck there. Sometimes for a really long time. 

In a first step, I added a timeout:

`final loc = await LocationManager().getLocation().timeout(Duration(seconds: 5));`

to make sure that the probe doesn't get stuck. But now I also added a dirty fix such that it actually uses Geolocator instead of Location on iOS:

```
Position p = await Geolocator.getCurrentPosition(
              timeLimit: Duration(seconds: 5));
```

This seems to work much better on iOS. I also don't use the `ContextSamplingPackage.CURRENT_LOCATION` on iOS because I also to find preform much worse on iOS compared to Android. ContextSamplingPackage.LOCATION seems to work somewhat better here (even though it also used the Location package as far as I know).

bardram, 2024-05-03:
Hm... I have been back and forth between using either the "location" or "geolocator" Flutter plugins many times. And it seems to change often which one is best. But let me have a look later.

bardram, 2024-05-05:
I've added a timeot to the getLocation() method. But why it doesn't work in the location plugin is a mystery.

bardram, 2024-05-05:
ok - couldn't make it work either - seems to be an issues with the location package. So changed (back) to using the geolocation package. 

Released as 1.6.0.

sc00n, 2024-06-17:
I fear that switching to geolocation may have been too quick. The coverage on iOS is really bad now. This is what I already found out earlier, that sensing without location really doesn't work (specifically on iOS). So the Location package does something so that the app can keep on sensing. Without the location package, you don't even get close to the frequency you want to sense on. Would it be possible to go back to the Location sensor? Additionally, the notification (which I really like) is gone on Android.

Coverage on Android seems to be good at first sight. 

My preferred option is to use the Location package for constant location sensing and the geolocation package for weather. I don't know if that is possible.

bardram, 2024-06-18:
ok - will reopen this issue and take a look.

bardram, 2024-06-19:
Seems to be a general problem with the location plugin - see issue [#916](https://github.com/Lyokone/flutterlocation/issues/916).

It seems like if there are no location updates, this method does not return.

bardram, 2024-06-19:
I've now changed back to using the `location` package. 

I've also re-implemented the `getLocation()` method, so it will return a location "most of the time". 

But I've noticed, that when debugging with the phone laying still on the table, you don't get location updates (and the `getLocation()` method hence times out). In order to ensure location updates even when the phone is laying still, change the `distance` filter to zero in the `LocationService`, like this:

```
    final locationService = LocationService(
      // used for debugging when the phone is laying still on the table
      distance: 0,
    );
    protocol.addConnectedDevice(locationService, phone);
````

closed 2024-06-19 by bardram