carp-dk/carp.sensing-flutter #390: step count doesnt work (anymore)

sc00n, 2024-05-02:
A few days before I didn't have any problems getting the step count. Now it doesnt come trough anymore. 

I tried the following:

```
    protocol.addTaskControl(
        ImmediateTrigger(),
        BackgroundTask()
          ..addMeasure(Measure(type: SensorSamplingPackage.STEP_COUNT)),
        phone);
```

and

```
    protocol.addTaskControl(
        PeriodicTrigger(period: Duration(seconds: 5)),
        BackgroundTask()
          ..addMeasure(Measure(type: SensorSamplingPackage.STEP_COUNT)),
        phone);
```

It works with 

```
  carp_core: 1.4.1
  carp_mobile_sensing: 1.4.7
```

but not with 

```
  carp_core: 1.5.0
  carp_mobile_sensing: 1.5.1
```

bardram, 2024-05-03:
Remember to add permission to access step counts / activity recognition API on both Android and iOS.

But - working on a new release and will check this again.

sc00n, 2024-05-06:
I update to 

```
carp_core: 1.6.0
  carp_mobile_sensing: ^1.6.0
  carp_context_package: ^1.6.0
```

But I still am not getting any stepcount (on iOS)

closed 2024-06-03 by bardram